ham-radio-software/D-Rats #2: Linux installation not possible with current setup script

yo2loj, 2020-04-16:
The current setup.py needed for easy linux installation is not up to date with the change in the 0.3.6:
- the script assume the main files are called "d-rats" and "d-rats_repeater", without the .py extension.
- the script tries to copy files from a non-existent share directory, which was present in 0.3.3 holding the user manuals, desktop icon and desktop shortcuts.

Workaround is a renaming of the main scripts, and import of the old share folder from 0.3.3.
So either a revert to the previous names is needed, or an update of the setup script, maybe even resurrection the share folder in the project.

Thank you for your efforts Maurizio!

maurizioandreotti, 2020-04-17:
HI Marius,  thanks

I am not familiar with easy linux installation, at this point in time I would prefer to keep .py extensions for d-rats and d-rats_repeater file. as changing this will imply reassessing the compilation procedure for win which is something i dont have time now.

can you propose an amend of the setup.py file for making it work on linux?

I will look to restore the share folder from 0.3.3.

thanks for you help!

yo2loj, 2020-04-17:
Ok, I will try to do the necessary updates to the script.

yo2loj, 2020-04-17:
Updated the needed files in pull request #4

yo2loj, 2020-04-17:
The installation steps are quite easy (assuming one has all the neede python libs installed):
- via package installer: python-glade2, python-libxml2, pythhon-libxslt1
- via pip: pyserial
After this, the steps are as follows:
- cd to your D-Rats source directory
- issue 'python setup.py build'
- issue 'python setup.py install'
This should do it. Main scripts are in /usr/local/bin, configuration and logs will be found in the user's home directory as .d-rats-ev  (hidden).

closed 2020-04-17 by maurizioandreotti

maurizioandreotti, 2020-04-17:
thanks! 
I add this indication in the Readme under linux install