ham-radio-software/D-Rats #14: D-RATS Shared folder issue

maurizioandreotti, 2020-04-20:
document how to solve the situations that sometimes happen about "D-RATS Shared" folder  and produce  pop-up window stating:

----
D-RATS has encountered an error.

This may be non-fatal, so you may click Ignore below to attempt to continue running.  Otherwise, click 'Quit' to terminate D-RATS now. If you are planning to file a bug for this issue, please click Debug Log below and include the contents in the bug tracker.

If you need to ignore all additional warnings for this session, click Ignore All.  However, please reproduce and report the issue when possible."
-----

AND creates lines in the debug log:
...
simplejson was not found. Geocoders relying on JSON parsing will not work.
FILE: C:\Users\USERNAME\AppData\Roaming\D-RATS\d-rats.config
Creating downlaod directory: C:\Users\USERNAME\Desktop\D-RATS Shared
---- GUI Exception ----
Traceback (most recent call last):
  File "d-rats", line 118, in <module>
  File "d_rats\mainapp.pyo", line 946, in __init__
  File "d_rats\config.pyo", line 1742, in __init__
WindowsError: [Error 3] The system cannot find the path specified: 'C:\\Users\\USERNAME\\Desktop\\D-RATS Shared'
---- End ----

maurizioandreotti, 2020-04-29:
from email by Woodrick, Ed : 
D-RATS not starting. When you look at the error, it indicates that the desktop folder can't be created. (only occurs on new startup). I've got it Noted on DSTARInfo.com. Can you change the initial startup code to either not create the Shared Folder at startup, or at least check and handle if the c"\users\username\Desktop folder is there?

maurizioandreotti, 2020-04-29:
d-rats doesn’t start for this reason why not create a “c:\ D-RATS Shared” folder and point to that one?

the path can be modified editing with wordpad the config file
C:\Users\YOUR USER \AppData\Roaming\D-RATS-EV\d-rats.config

And changing this line pointing to a different folder in[prefs] the voice:

download_dir = C:\Users\mauri\Desktop\D-RATS Shared
download_dir = C:\D-RATS Shared

kelvin0mql, 2021-02-22:
I realize **this** issue is one of Documentation, but nonetheless it is an open issue about the **Shared** dir, and it touches on something that I may even be so crazy as to clone and try to change for my own needs.

https://groups.io/g/d-rats/topic/feature_suggestion_for_the/80825576

I think the path to which I download (and other stations upload to me) should not necessarily be the same path from which others can download from me. 

i.e. Not everything that I receive is something I **should Share**, necessarily. I may need these to be distinct decisions.

wb8tyw, 2023-04-03:
This is a duplicate of https://github.com/ham-radio-software/D-Rats/issues/149