ham-radio-software/D-Rats #25: RS-MS1A INTEGRATION

maurizioandreotti, 2020-04-23:
Icom relaesed an android app RS-MS1A which could be integrated with D-Rats

Sample text messages received by:
[09:38:45] [com6] CQCQCQ: $$Msg,IZ2LXI,,0011DCtest transmission

messages sent by an initial version of RS-MS1A were reported as "received" by D-Rats, 
messages sent by D-Rats to RS-MS1A were reported as not seen on RS-MS1A

yo2loj, 2020-04-23:
I assume it just puts the radio in external gw terminal mode and forwards the data port, since the cable takes up the single USB port on the handheld.
But it is a nice feature.

maurizioandreotti, 2020-05-07:
putting  a reference to the code:
https://github.com/maurizioandreotti/D-Rats/blob/master/d_rats/mainwindow.py

line 187

                # original d-rats string 
                self.emit("user-send-chat", "CQCQCQ", port, "?D*%s?" % d, True)
                
                #sample rs-ms1a string CQCQCQ: $$Msg,IZ2LXI,,0011DCtest transmission
                #self.emit("user-send-chat", "CQCQCQ", port, "$$Msg,IZ000,,0011D,%s" % d, True)

wb8tyw, 2023-04-06:
The RS-MS1A has a feature for picture file transfer also.  We need to get some information on how it differentiates between a picture and text in the data stream.

We need to know how binary data is handled in the pictures and if any compression is used.

wb8tyw, 2023-09-22:
While it may appear that RS-MS1A and D-Rats send plain text, they do not.
Some of the 8 bit codes, or octets that are sent are interpreted by the radio it self and not sent over the air when sent to the radio from the serial port.

D-Rats uses y-encoding / y-decoding to translate these octets so that they show up on the distance D-Rats system as originally sent.

RS-MS1A needs to be doing the same thing, but may not be using y-encoding to handle it.

So we need to find out how RS-MS1A handles these special octets, and how RS-MS1A differentiates sending text and pictures,

wb8tyw, 2024-06-09:
Text sample:  ‘$$Msg,WB8TYW,CQCQCQ,0011ABfooD’
What we don't know:
- How to generate the what in the above example is "0011AB'.
- How to generate for the final character after the body of text of the message.
- What characters are allowed in the message.

For the above message I was able to get RS-MS1A to accept it when sent to it from a Windows 10 system via a BlueTooth connection.

wb8tyw, 2024-06-09:
Sample picture sent at the lowest resolution/quality was sent as 80 tiles in a 10x8 array.   I have not tried other settings.
This was just the rendered text on the terminal, and I expect that there are a lot of non-printing characters and also lines that were overwritten.

```text
$$Pic,WB8TYW,,018E▒▒▒▒▒▒1 v▒▒[▒-▒8▒▒▒]Ԯ
$$Pic,WB8TYW,,018E▒▒NMK▒W▒U▒▒j(▒▒▒g9▒▒!h▒G▒▒e▒&
$$Pic,WB8TYW,,018E▒▒▒▒▒6PG"▒▒▒▒?pn⠳▒
$$Pic,WB8TYW,,018E▒▒▒▒>▒▒▒a%▒▒▒)▒▒S▒▒▒J▒4{▒}▒▒
$$Pic,WB8TYW,,018E▒▒рsP\▒m▒▒▒▒H▒▒QI▒h▒▒▒▒O֦▒\▒?]
$$Pic,WB8TYW,,018E▒▒▒▒▒#▒g▒X
$$Pic,WB8TYW,,018E▒▒Z▒▒     6~▒y2▒\▒=▒64h▒
$$Pic,WB8TYW,,018E▒▒ ▒▒▒▒▒ ▒g▒d▒O▒
$$Pic,WB8TYW,,018E▒▒eFi%gD(x▒▒▒/8▒▒▒Fx▒*
$$Pic,WB8TYW,,018E▒▒▒"▒^▒̤▒▒▒▒qՕqP▒▒▒"▒▒▒m▒o▒▒&[?▒
$$Pic,WB8TYW,,018E▒▒▒▒h3;&▒▒7▒.▒4?.B▒▒O▒▒-▒Y▒
$$Pic,WB8TYW,,018E▒X▒8▒▒▒Lrt<▒?▒▒H&▒▒U▒!▒I▒g=▒▒M▒
$$Pic,WB8TYW,,018E▒▒9J▒▒s▒▒▒▒b8▒Y▒Թ▒ǳ▒▒▒5QA▒Z9▒@▒▒▒P3
                    ▒T▒R
$$Pic,WB8TYW,,018E▒}Q▒N▒▒▒▒!▒E▒▒528۵z{r08▒▒\)n▒▒G
```

wb8tyw, 2024-07-14:
As we could be getting MS1A messages and D-RATS messages on both ratflectors and radio links, we need to add more information for the stations being heard to track which protocol the station was heard using to be able to have the option to respond to a station in the format that we got a message from it.