ham-radio-software/D-Rats #27: invocation of firefox and gedit from linux

maurizioandreotti, 2020-04-23:
add in config a parametrer to point to browser and text editor (currently firefox and gedit are hardcoded) used by execution on linux to show log and other stuff on internet (i.e. lets not take for granted that all users will have these programs)

wb8tyw, 2023-04-06:
The GUI packages for most Linux systems and for Microsoft Windows allows the setting of the default programs to do functions like text editing and web browsing.

If we set D-rats to just use GTK to launch an application to to the function, that is less platform dependent code we need to maintain.

We may want to add a feature to allow users to select a program to override GTK or for the cases were GTK is unable to find a program.

wb8tyw, 2023-06-04:
Adding a Pull Request to have D-Rats use the default programs for text and html file viewing.

More advanced configuration programming can be done in the future to add an option to specify which application to use.

wb8tyw, 2023-06-25:
Now for Windows and Linux, the application configured by the GUI should be used.
https://github.com/ham-radio-software/D-Rats/issues/196 which is a MAC OS specific problem outside of D-Rats is blocking fixing this completely.