ham-radio-software/D-Rats #30: Windows - installer

maurizioandreotti, 2020-04-23:
create windows installer vs the rar file?

yo2loj, 2020-04-23:
Create single file in py2exe?

yo2loj, 2020-04-23:
BTW, maybe zip is a better option. Windows can unpack zip files by default, rar needs an 3rd party program.

maurizioandreotti, 2020-04-23:
long story about zip...

> Not a bad idea, but is it really needed?...

maybe not, and could keep the users "less aware" ... let's keep this for later

dranch, 2020-04-24:
If there was a proper installer, all of this "which version of this .dll do you have" and "oh, you're missing this LZH file", etc. all goes away.  Yes, running a program by just decompressing and running it is easy but is running an installer really that hard?

maurizioandreotti, 2020-04-24:
Sounds nice, I fear that it (the installer) will become another piece to take care

 

Btw could you provide one?

dranch, 2020-04-24:
Sorry, I generally don't do Windows so I don't have much to offer here.  There are various open source installers out there for Windows but I'm sure they aren't all equal in terms of support, features, etc.  Sorry I can't be more help here.