ham-radio-software/D-Rats #31: fix winlink sending email lock

maurizioandreotti, 2020-04-23:
In the branch of 0.3.7 with merged code from 2012 which was (also) stated to implement winlink refinements I am now able to receive email, but when i send i get a file block issue - these are 3 emails blocked.

Msgrouting: Running routing loop
Msgrouting: ------ LOCK OWNED BY -------
  File "/usr/lib/python2.7/threading.py", line 774, in __bootstrap
    self.__bootstrap_inner()
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/d-delta-master/d_rats/msgrouting.py", line 520, in _run
    queue = self._get_queue()
  File "/home/pi/d-delta-master/d_rats/msgrouting.py", line 273, in _get_queue
    if not msg_lock(f):
  File "/home/pi/d-delta-master/d_rats/msgrouting.py", line 63, in msg_lock
    traceback.print_stack(file=lf)
------------

Msgrouting: Message /home/pi/.d-rats-ev/messages/Outbox/form_04232020_160625.xml is locked, skipping
Msgrouting: ------ LOCK OWNED BY -------
  File "/usr/lib/python2.7/threading.py", line 774, in __bootstrap
    self.__bootstrap_inner()
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/d-delta-master/d_rats/msgrouting.py", line 520, in _run
    queue = self._get_queue()
  File "/home/pi/d-delta-master/d_rats/msgrouting.py", line 273, in _get_queue
    if not msg_lock(f):
  File "/home/pi/d-delta-master/d_rats/msgrouting.py", line 63, in msg_lock
    traceback.print_stack(file=lf)
------------

Msgrouting: Message /home/pi/.d-rats-ev/messages/Outbox/form_04232020_160741.xml is locked, skipping
Msgrouting: ------ LOCK OWNED BY -------
  File "/usr/lib/python2.7/threading.py", line 774, in __bootstrap
    self.__bootstrap_inner()
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/d-delta-master/d_rats/msgrouting.py", line 520, in _run
    queue = self._get_queue()
  File "/home/pi/d-delta-master/d_rats/msgrouting.py", line 273, in _get_queue
    if not msg_lock(f):
  File "/home/pi/d-delta-master/d_rats/msgrouting.py", line 63, in msg_lock
    traceback.print_stack(file=lf)
------------

maurizioandreotti, 2020-04-23:
@yo2loj ...if we can fix and check all parts of winlink (attachments and RMS) maybe we can then proceed sharing a 0.3.7 compiled version to all... as of today i finished petrol on this merge

yo2loj, 2020-04-23:
I have seen this on the old one... It happens becoause on message creation sometimes the removal of the lock is somehow skipped. Will check.

yo2loj, 2020-04-23:
Fixed in #32. Merged.

yo2loj, 2020-04-23:
Not completely fixed... Next iteration needed...

yo2loj, 2020-04-24:
Now all messages with and without attachment, send and received work OK.
Merged #33

This issue can be closed.

closed 2020-04-24 by maurizioandreotti