ham-radio-software/D-Rats #39: MAC version.....?

maurizioandreotti, 2020-04-24:
in theory it should be possible to run D-Rats on Mac, but I have no Mac to test if it's still possible to execute and compile...

yo2loj, 2020-04-24:
Sorry, I can not help on this one...
But just as a hint, WL2K will not work out of the box, since LZHUF is not available on the Mac.
But since we have the source code now... Let's check :-)

wb8tyw, 2021-12-30:
lzhuf is locally built for d-rats for any platform that it is missing on.

look at the libexec directory in the source.

wb8tyw, 2022-01-02:
Module wl2k.py function run_lzhuf has code to disable when running on darwin.
If the lzhuf package is locally built and placed in the right location, it appears that commenting out that code may be all that is needed to make it work.

I do not know why that comment is present.
As near as I can tell, the lzhuf package is no longer provided for any platform.
Can someone work with the wl2k developers to add support for a more widely used compression/decompression library?

wb8tyw, 2023-04-12:
[LZHUF](https://github.com/ham-radio-software/lzhuf) is now in its own project.  Mac OS-X users will need to locally build it and install it as per the instructions.

The lzhuf that was formerly built into D-Rats did not work on big-endian systems, which the Mac systems based on PowerPC, and Solaris/SPARC systems are.

File transfers on D-Rats older than the 0.4.0 version on Big Endian systems can not transfer files to Little Endian systems such as Intel based systems.

That saying it would really help if we could get someone to build Mac based packages of D-Rats.

closed 2023-04-12 by wb8tyw