ham-radio-software/D-Rats #40: chat - Send to all ports

maurizioandreotti, 2020-04-27:
In the chat view, is there a way to send a message to all remote RAT servers or do I have to manually select one, send my message, manually select the next one, send the same message, etc?

To my knowledge it shall be done one by one.  If I recall well the online/attended/unattaneded already goes to all ports, so adding an “all ratflectors” option could be developed, not really sure how much useful vs possible risk of misuse.

dranch, 2020-04-27:
The "ping all" stations feature available as a right-click in the right-hand station view is similar to what I'm talking about here.  Others have mentioned to use the QST feature but I feel that's not suitable for chat.  My use case here is the following:

Let's say I'm connected to FIVE different ratflectors and I have a question that I'd like to get answered.  I'd like to have an "All RATs" in the lower left chat view to send my question to.  My question would then flood out and the UI could then flip back to the first entry in the list.  if I get a response from someone on a specific RAT, I could then change the RAT to the one the response came from

wb8tyw, 2023-04-05:
I am not sure the d-rats implementation of ping-all is robust enough to deal with radio connected d-rats clients.
I am not sure how well the d-rats deals with collision avoidance.

Related:
- https://github.com/ham-radio-software/D-Rats/issues/73
- https://github.com/ham-radio-software/D-Rats/issues/74