ham-radio-software/D-Rats #42: Config --> Preferences --> Radio -- Check on Ratflector status

maurizioandreotti, 2020-04-27:
In the Config --> Preferences --> Radio view, would it be possible to add an indicator of which connections are ACTIVE?   This would be an easy way to determine which remote Ratflectors are working or not.

you mean a sort of a ping to check the ratflector before marking it as active?

dranch, 2020-04-27:
If you look at the D-rats "event log" tab, you can see which remote RAT hosts accept or refuse connections but it's tedious to see what stations work.  There also isn't a clear association of what hostname corresponds to which alias name in the the Event log.

drats.ratflector.com - connection timed out
ref.d-rats.com - connection refused
k3pdr rat - connected
wa7fw.dstargateway.org - connected

wb8tyw, 2023-04-05:
This is essentially a duplicate of https://github.com/ham-radio-software/D-Rats/issues/74 which has more discussion on it.

closed 2023-04-05 by wb8tyw