ham-radio-software/D-Rats #43: D-rats wants to use the legacy OSS system on Linux

maurizioandreotti, 2020-04-27:
D-rats wants to use the legacy OSS system on Linux.  This sound system needs to be updated.  The easiest way might be to just allow system commands to run a program with a specified file.  At that point, I could use Alsa's "aplay" with the file.
--
dPlatform :Error playing sound /usr/share/sounds/purple/receive.wav: [Errno 2] No such file or directory: '/dev/dsp'                                                                                           
dPlatform :Error playing sound /usr/share/sounds/purple/receive.wav: [Errno 2] No such file or directory: '/dev/dsp'                                                                                           
dPlatform :Error playing sound /usr/share/sounds/purple/receive.wav: [Errno 2] No such file or directory: '/dev/dsp'

dranch, 2020-07-24:
Sound still wasn't working for me even with the newest 0.3.10Beta2 version but on a lark, I did some OSS troubleshooting and it seems that Linux wasn't loading the OSS kernel modules.  You can tell as none of the /dev/dsp* devices were listed.  I ran "sudo modprobe snd_pcm_oss" and then D-rats sound started working!  That's great but I notice that when D-rats plays a sound, there is a loud popping sound at the beginning which is pretty annoying.  I don't hear this sound when I try things manually from the command prompt with say "aplay /usr/share/sounds/purple/receive.wav" or even "aoss aplay /usr/share/sounds/purple/receive.wav".  Any ideas where the popping sound would be coming from?  

Regardless, I think that if the code could be easily changed to just make a OS system call to "aplay" on Linux machines, this will remove the OSS dependency and be a lot more optimal.  Maybe add to the D-Rats sound settings area to have a variable for what full path and command line audio player tool the user wishes to use".  With these two options, then D-rats could play MP3s, Oggs, etc.  Whatever the users wants.

wb8tyw, 2023-04-05:
Before adding a system call, check to see if there is a GTK related library to do the operation.  Some of the stuff in dplatform.py that is not 100% can be probably be replaced with GTK calls, and simplify our codebase.

wb8tyw, 2023-06-04:
Fixed with https://github.com/ham-radio-software/D-Rats/pull/192

closed 2023-06-04 by wb8tyw