ham-radio-software/D-Rats #44: mainapp.py:1245: GtkWarning

maurizioandreotti, 2020-04-27:
these GTK warning are there since ever I think.  I am not sure if it’s worth investing time at the moment  - if anybody has a fix, more then welcome

mainapp.py:1245: GtkWarning: Failed to set text from markup due to error parsing markup: Error on line 3: Entity did not end with a semicolon; most likely you used an ampersand character without intending to start an entity - escape ampersand as &amp;                                                   
  gtk.main()

wb8tyw, 2023-04-05:
Have not seen them in the Python3 version.
These GTK Warnings are hard to chase down as they usually do not give enough context to find their source in D-Rats, and some are not even from D-Rats, but from third party libraries we have no control over.

wb8tyw, 2023-04-05:
If you are not seeing these warnings with the Python3 version, can you close this bug?

maurizioandreotti, 2023-04-06:
vclosing as the warning are not showing anymore

closed 2023-04-06 by maurizioandreotti