ham-radio-software/D-Rats #47: Upgrate

maurizioandreotti, 2020-04-28:


maurizioandreotti, 2020-04-28:
...what shall be done to port D-Rats to new version of Python and GTK and also get rid of all libraries not supported anymore?

I am placing this question to get help start understanding 
* is it worth? 
* what is the size of what shall be done? 
and 
- *an this be split in manageble chunks ?

dranch, 2020-04-28:
I would look to see what's being done with the Chirp project to get some ideas of what to move to. There are some tools out there that helps some level of Python2 to Python3.3+ porting but the GUI change might require the most manual work.

maurizioandreotti, 2020-04-29:
yes... going through the code i understand that the interface events guide a good part of the program behaviour, so not sure if it is needed to have an intermediate release adding a layer to then disconnect the GUI and connect a new one

maurizioandreotti, 2020-04-29:
https://stxnext.com/blog/2019/04/16/python-3-migration-guide/#first-steps... 

in spare time i am doing an intial step locally with python-modernize - if it works ... will be a first step 0.3.8

vishwin, 2020-06-08:
> I would look to see what's being done with the Chirp project to get some ideas of what to move to. 

CHIRP's py3 branch switched wholesale to wxPython 4, which is only one of several options when it comes to GUI toolkits with Python bindings. PyGTK has been abandoned since 2011; the successor has been accessing the GObject bits of GTK directly through PyGObject 3.

closed 2023-04-05 by wb8tyw