ham-radio-software/D-Rats #48: 0.3.8 beta modernize dont start after compiled on XP -

maurizioandreotti, 2020-04-30:
Mainapp   : Enabled debug log for Win32 systems
Traceback (most recent call last):
  File "d-rats.py", line 127, in <module>
  File "d_rats\mainapp.pyo", line 60, in <module>
  File "d_rats\mainwindow.pyo", line 46, in <module>
  File "d_rats\ui\main_chat.pyo", line 34, in <module>
  File "d_rats\qst.pyo", line 37, in <module>
  File "d_rats\mapdisplay.pyo", line 39, in <module>
  File "d_rats\geocode_ui.pyo", line 24, in <module>
  File "d_rats\geopy\__init__.pyo", line 2, in <module>
  File "d_rats\geopy\geocoders.pyo", line 7, in <module>
  File "six.pyo", line 203, in load_module
  File "six.pyo", line 115, in _resolve
  File "six.pyo", line 82, in _import_module
ImportError: No module named xmlrpclib

as of now i am not able to find thislibrary to install on XP

yo2loj, 2020-04-30:
This library has been renamed to xmlrpc.client in Python 3

maurizioandreotti, 2020-04-30:
yes, you're right. modernize changed geopy to import that from six:
import six.moves.xmlrpc_client

I have easy installed six on the XP but i get that error tryng to execute the exe

i am still compiling on 2.7, exactly as before. just run the code through modernize to have the automatic conversions and start moving to py3. I dont tthink it will work already on py3... but i have not yet tested 
 
btw to now the only real code change was about the code to save pngs from urls for the map tiles and some issues on multiple import of same d-rats modules (I commented out the unneeded import lines with #py3),  on linux raspi still works as before, ...

...on XP (i just compiled a couple of times) i get that error launching the exe (same issue executing the exe on win10). 

really XP .... I have not really yet had time to invest on compiling fo widows. just used the XP VM machine which i know it was working before the "modernize".
also the XP VM it's just a clean install of windows that i had devoted to d-rats compile at beginning of covid shutdown to compiled-rats for windows - i dont want to mess up my win10 work pc (again) ... and dont have time to setup a dedicated one... 

if you have a bit of time can you try a compile on 7 (or on py3) and see if it produces some nicer result?

maurizioandreotti, 2020-05-03:
so... i got to understand a bit more:

the modernized code (which should be a step forward to be compatible with python3) runs on XP if launched via cmd like "python2 d-rats.py" but it doesnt start as compiled because the following libraries are not copied into the library.zip:
- xmlrpclib.pyo
- SimpleXMLRPCServer.pyo
- SocketServer.pyo
- BaseHTTPServer.pyo

copying manually into library.zip solves (for now)

next step it's to undertand why are not copied by the config.py script

closed 2020-05-03 by maurizioandreotti