ham-radio-software/D-Rats #52: Unneeded files in the git repository

yo2loj, 2020-05-05:
Please take care not no place system specific files in the repository, since they get created on build:
- in libexec: *.o, lzhuf (the linux binary and object files)
- in share: *.1.gz (compressed manual pages), mapdownloader.desktop (not available anymore)

The best solution is to add these files to gitignore, so they don't appear again.

yo2loj, 2020-05-05:
Solved in #53

closed 2020-05-05 by yo2loj

maurizioandreotti, 2020-05-05:
Hi, after pulling in local the last  commit to remove uncessary files i get the followin error related to missing lzhuf.

I guess it would be sorted out making linux install of d-rats once downloaded?

for me, I can fix re-adding locally the lzhuf locally where i need it to develop, but i think this will create questions from all people that currently use d-rats simply downloading the code and execute it.... i fear people will ask and need to be supported 

wl2k       : Found data at 37
wl2k       : Reading data block 250 bytes
wl2k       : Reading at 289
wl2k       : Found data at 289
wl2k       : Reading data block 49 bytes
wl2k       : Reading at 340
wl2k       : Found eof at 340
wl2k       : Got data: 299 bytes
Exception in thread Thread-10:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/home/pi/D-Rats-Ev-Master/d_rats/wl2k.py", line 605, in run
    result = self._run_incoming()
  File "/home/pi/D-Rats-Ev-Master/d_rats/wl2k.py", line 564, in _run_incoming
    count = wl.get_messages()
  File "/home/pi/D-Rats-Ev-Master/d_rats/wl2k.py", line 374, in get_messages
    msg.read_from_socket(self._conn)
  File "/home/pi/D-Rats-Ev-Master/d_rats/wl2k.py", line 251, in read_from_socket
    self.__content = self.__decode_lzhuf(data)
  File "/home/pi/D-Rats-Ev-Master/d_rats/wl2k.py", line 118, in __decode_lzhuf
    return run_lzhuf_decode(data)
  File "/home/pi/D-Rats-Ev-Master/d_rats/wl2k.py", line 83, in run_lzhuf_decode
    return run_lzhuf("d", data[2:])
  File "/home/pi/D-Rats-Ev-Master/d_rats/wl2k.py", line 66, in run_lzhuf
    shutil.copy(os.path.abspath(lzhuf_path), cwd)
  File "/usr/lib/python2.7/shutil.py", line 139, in copy
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 96, in copyfile
    with open(src, 'rb') as fsrc:
IOError: [Errno 2] File o directory non esistente: '/home/pi/D-Rats-Ev-Master/libexec/lzhuf'

yo2loj, 2020-05-05:
Under linux, the code for lzhuf is built when doing 'python setup.py build' and is specific for each computer architecture, and then it is copyed in the proper directory during install. But for this, you need a working gcc environment installed.
It is not acceptable to put a specific version of the file in the repository and then expect it to work on every machine. The only option is to build it on the target machine.