ham-radio-software/D-Rats #60: install on linux - gladexml object error

maurizioandreotti, 2020-05-07:
Hi Marius,
i bumped into some issue on linux instal which i dont understand - i guess some library is not copied in the right path or just need a new reference after the modernization... hope you can help here.

with 0.3.8 code, (which runs if launched by ./d-rats.py locally)  after executing the linux install procedure, launching d-rats from the install dir, I get this error:

Mainapp   : load main window with self config
Permission deniedPermission deniedI/O warning : failed to load external entity "/usr/share/d-rats/ui/mainwindow.glade"

(d-rats.py:3347): libglade-WARNING **: 23:45:52.437: document not well formed
---- GUI Exception ----
Traceback (most recent call last):

  File "/usr/local/bin/d-rats.py", line 136, in <module>
    app = mainapp.MainApp(safe=opts.safe)

  File "/usr/local/lib/python2.7/dist-packages/d_rats/mainapp.py", line 1120, in __init__
    self.mainwindow = mainwindow.MainWindow(self.config)

  File "/usr/local/lib/python2.7/dist-packages/d_rats/mainwindow.py", line 286, in __init__
    "mainwindow", "D-RATS")

RuntimeError: could not create GladeXML object

---- End ----

maurizioandreotti, 2020-05-07:
ps - i did both "setup.py build" and "setup.py install"
 with sudo....

wb8tyw, 2023-04-05:
Closing as this is for the now unsupported python2 branch.

Using sudo with pip or setup.py is strongly discouraged, and can leave your python installation broken.

closed 2023-04-05 by wb8tyw