ham-radio-software/D-Rats #74: Display connected status of remote Ratflectors

dranch, 2020-06-11:
It would be very helpful to add a connected/disconnected indicator under Config --> Radio for various ratflectors.  Maybe a green vs. blue vs "white fill in a circle"  indicator in column to the right of the "port" name to indicate connected, connecting, or disconnected.

It would be good to also know if D-rats does any periodic retries if it couldn't connect to the ratflector?  It might be too complex / UI overload to show a counter of connection attempts per ratflector.

dslotter, 2020-06-11:
Wish there were a voting or "like" button for what David mentioned above.

wb8tyw, 2023-04-05:
D-Rats tries to connect a few times to a ratflector until it gives up.
Entering the config radio section and then saving will start it trying to connect to the ratflectors.
Python 3 version will attempt to reconnect to a ratflector after it has once connected to it successfully and then lost the connection.  However detecting some disconnections seems to have a problem.

wb8tyw, 2023-04-05:
I think I would like a way to see the ratflector status with out having to go into the config screen.   Need to think about the best way to handle the display.

Related: https://github.com/ham-radio-software/D-Rats/issues/70