ham-radio-software/D-Rats #75: Automatic retry and reconnection to remote D-rats ratflector

dranch, 2020-06-22:
I connect to a remote ratflector via 127.0.0.1:9000 via SSH portfw.  I've noticed that if the remote ratflector becomes unavailable for a short period, my local D-rats client will get disconnected and won't try to re-connect.  I believe D-rats should try to reconnect to any configured and *enabled* Ratflector say every 60 seconds.

wb8tyw, 2022-09-08:
Originally seen on Microsoft Windows, but also seen occasionally on Anti-X linux.

Sometimes the D-Rats software stops getting updates from the Internet, but does not detect that it is disconnected from the ratflector.

It looks like a disconnected port returns zero bytes read, and since we are using a 1/2 duplex system for TCP connections, it looks like we sometimes can not see the difference between a broken connection and no data available for receipt.

wb8tyw, 2023-04-05:
Many ISPs and possibly home firewall routers will disconnect TCP sessions that do not transfer data over a period of time.

The only way to prevent the disconnection is to setup a timed QST, which could be annoying to some hams.

Related: https://github.com/ham-radio-software/D-Rats/issues/134