ham-radio-software/D-Rats #76: message reception with 0.3.10beta1 is broken and results in 0 byte files that crashes the D-rats client

dranch, 2020-07-12:
With 0.3.10beta1, when receiving a message and then navigating to the inbox, I get the below error and in my messages directory, it creates 0 byte files which freaks out D-rats".  Downgrading to 0.3.9 Final works fine.
--
~/.d-rats-ev/messages/Inbox$ ls -la
total 16
drwxrwxr-x 2 dranch dranch 4096 Jul 12 15:20 .
drwxrwxr-x 7 dranch dranch 4096 Apr  4 08:28 ..
-rw-rw-r-- 1 dranch dranch  908 Jun 15 10:32 form_06152020_103253.xml
-rw-rw-r-- 1 dranch dranch  499 Jun 18 09:51 form_06182020_095124.xml
-rw-rw-r-- 1 dranch dranch    0 Jul 12 14:13 form_07122020_141317.xml
-rw-rw-r-- 1 dranch dranch    0 Jul 12 14:18 form_07122020_141756.xml
-rw-rw-r-- 1 dranch dranch    0 Jul 12 15:09 form_07122020_150927.xml
--

--
/home/archive/D-rats/D-Rats-0.3.9-final/d_rats/mainwindow.py:286: GtkWarning: IA__gtk_notebook_set_tab_label: assertion `GTK_IS_WIDGET (child)' failed                                                         
  "mainwindow", "D-RATS")                                                                                                                                                                                      
---- GUI Exception ----                                                                                                                                                                                        
Traceback (most recent call last):                                                                                                                                                                             

  File "/home/archive/D-rats/D-Rats-0.3.9-final/d_rats/ui/main_messages.py", line 1118, in <lambda>
    lambda x, y: self._messages.open_folder(y))                                                    

  File "/home/archive/D-rats/D-Rats-0.3.9-final/d_rats/ui/main_messages.py", line 714, in open_folder
    self.refresh()                                                                                   

  File "/home/archive/D-rats/D-Rats-0.3.9-final/d_rats/ui/main_messages.py", line 701, in refresh
    self._update_message_info(iter)                                                              

  File "/home/archive/D-rats/D-Rats-0.3.9-final/d_rats/ui/main_messages.py", line 662, in _update_message_info
    form = formgui.FormFile(fn)                                                                               

  File "/home/archive/D-rats/D-Rats-0.3.9-final/d_rats/formgui.py", line 618, in __init__
    raise Exception("Form file %s is empty!" % filename)                                 

Exception: Form file /home/dranch/.d-rats-ev/messages/Inbox/form_07122020_153021.xml is empty!

---- End ----
--

maurizioandreotti, 2020-07-14:
solved with https://github.com/maurizioandreotti/D-Rats/releases/tag/0.3.10beta2

closed 2020-07-14 by maurizioandreotti

dranch, 2020-07-14:
Thank you for the quick turn around.  I'll try to validate