ham-radio-software/D-Rats #77: Persistent chat buffers when remote stations re-connect

dranch, 2020-07-13:
One really great feature in modern communications clients like Slack, Teams, etc. is when you connect to a chat, you can see all of the *previous* chat messages.  Having something like this would be hugely helpful when stations temporarily loose connection and then re-establish yet they missed various chats.  I imagine this "previous" state would be best offered via an enhanced D-rats Ratflector and would need to be configurable so it cannot consume too much resources on the Ratflector.

maurizioandreotti, 2020-07-14:
In the meantime have a look here on what is already there about chat messages logs in D-Rats client.

https://github.com/maurizioandreotti/D-Rats/wiki/111-How-to-get-old-chat-messages-from-the-logs

dranch, 2020-07-14:
Ah, having users look at C:\Users\YOUR USER NAME\AppData\Roaming\D-RATS-EV\logs will be helpful for the messages they previously received.  I'm looking for a way to catch back up on messages we NEVER received when the client was restarting.  Some of these modern clients fill in the missing blanks as you scroll back in the buffer.  Kinda slick.

wb8tyw, 2023-04-04:
Need some more information here on what is wanted.
Internet chat systems store the remote messages on a central server so that they can catch up.

D-Rats does not have a central server.  The ratflectors do not store any of the messages they sent.  In short, if a message is missed by a client, there is no way for a client to retrieve it.

As near as I can tell, there is limited communications with the Ratflector after logging in as far as sending and receiving messages.  So I am also not sure how hard it will be to modify the ratflector to store messages for catch up and have a way to retrieve them.  I am still working on mapping the D-Rats protocols, and I do not remember finding something that can indicate either a protocol version, or a way of getting non-message data from a Ratflector.

I have not looked at how D-Rats logs past messages it received to see if it is practical to have a way for D-Rats to display them.