ham-radio-software/D-Rats #79: Large amount of map data being left in /tmp

dranch, 2020-07-29:
On D-rats 0.3.10Beta2, I'm seeing large amounts of map data being left over in /tmp.  These left over files seem to go in triples created day consuming about 163MB.  I feel that one new files are downloaded, the old files should be deleted.  

Today's example:
-rw-------   1 dranch dranch  29392896 Jul 29 11:34 tmpXRENJd                                                   
-rw-------   1 dranch dranch     35205 Jul 29 11:05 tmpYqavPZ                                                   
-rw-------   1 dranch dranch 133724837 Jul 29 11:05 tmp1PzF37 

The file 
-rw-------   1 dranch dranch  29392896 Jul 29 11:34 tmpXRENJd
has the top contents:
--
# ---------------------------------- WARNING ----------------------------------------
# Some of the data that you have obtained from this U.S. Geological Survey database
# may not have received Director's approval. Any such data values are qualified
# as provisional and are subject to revision. Provisional data are released on the
# condition that neither the USGS nor the United States Government may be held liable
# for any damages resulting from its use.
#
# Additional info: https://help.waterdata.usgs.gov/policies/provisional-data-statement
#
# File-format description:  https://help.waterdata.usgs.gov/faq/about-tab-delimited-output
# Automated-retrieval info: https://help.waterdata.usgs.gov/faq/automated-retrievals
#
# Contact:   gs-w_support_nwisweb@usgs.gov
# retrieved: 2020-07-29 14:05:09 EDT       (sdww02)
#
# Data for the following 21914 site(s) are contained in this file
#    USGS 01010000 St. John River at Ninemile Bridge, Maine
#    USGS 01010070 Big Black River near Depot Mtn, Maine
#    USGS 01010500 St. John River at Dickey, Maine
#    USGS 01011000 Allagash River near Allagash, Maine
#    USGS 01011500 St. Francis River near Connors, New Brunswick
.
.
.
--

The file 
-rw-------   1 dranch dranch     35205 Jul 29 11:05 tmpYqavPZ                                                   
has the contents
--
<!DOCTYPE html>
<html lang="en">
<head>
<title>USGS User Input Message</title>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">

<meta name="viewport" content="width=device-width,initial-scale=1"/>
<meta http-equiv="P3P" content='policyref="https://waterdata.usgs.gov/w3c/p3p.xml"'>
<link rel="P3Pv1" href="https://waterdata.usgs.gov/w3c/p3p.xml" type="text/xml" />
<link rel="alternate" type="application/rss+xml" title="USGS Water Data for the Nation and USGS Water Services News" href="https://help.waterdata.usgs.gov/news/RSS/" />
<link rel="stylesheet" type="text/css" href="/nwisweb/stylesheets/USGS_common.css" />
<link rel="stylesheet" type="text/css" href="/nwisweb/stylesheets/USGS_custom.css" />

<link rel="stylesheet" media="screen and (min-width: 180px) and (max-width: 450px)" href="/nwisweb/stylesheets/nwiswebfluidsmallest.css" />
<link rel="stylesheet" media="screen and (min-width: 450px) and (max-width: 900px)" href="/nwisweb/stylesheets/nwiswebfluid.css" />
<link rel="stylesheet" media="screen and (min-width: 900px)" href="/nwisweb/stylesheets/nwisweb.css" />
.
.
.
--
 
The file 
-rw-------   1 dranch dranch 133724837 Jul 29 11:05 tmp1PzF37 
has the contents:
--
# ---------------------------------- WARNING ----------------------------------------
# Some of the data that you have obtained from this U.S. Geological Survey database
# may not have received Director's approval. Any such data values are qualified
# as provisional and are subject to revision. Provisional data are released on the
# condition that neither the USGS nor the United States Government may be held liable
# for any damages resulting from its use.
#
# Additional info: https://help.waterdata.usgs.gov/policies/provisional-data-statement
#
# File-format description:  https://help.waterdata.usgs.gov/faq/about-tab-delimited-output
# Automated-retrieval info: https://help.waterdata.usgs.gov/faq/automated-retrievals
#
# Contact:   gs-w_support_nwisweb@usgs.gov
# retrieved: 2020-07-29 11:02:09 EDT       (sdww01)
#
# Data for the following 21914 site(s) are contained in this file
#    USGS 01010000 St. John River at Ninemile Bridge, Maine
#    USGS 01010070 Big Black River near Depot Mtn, Maine
#    USGS 01010500 St. John River at Dickey, Maine
#    USGS 01011000 Allagash River near Allagash, Maine
#    USGS 01011500 St. Francis River near Connors, New Brunswick
#    USGS 01012960 Fish River at Soldier Pond, Maine
#    USGS 01013500 Fish River near Fort Kent, Maine
.
.
.
--

My configuration in D-rats is the following:
   Preferences --> Maps:
      - Map to use: Outdoors
      - Outdoors server URL: https://tile.thunderforest.com/outdoors/
      - Key string for Outdoors URL: <empty>
      - Freshen map after: 720

maurizioandreotti, 2021-10-08:
from the reported text i dont really get what data is that-
 is it about jpg tiles of map download or something else?
what happens in deleting the files with r,-f? does it regrow?

wb8tyw, 2021-12-30:
The USGS has totally changed their URLs and download API from what D-RATS can use.

So instead of downloading MAP data from pre-configured maps, you are getting an HTML error messages instead.  Something is probably not cleaning that up.

The new API appears to require a LAT/LONG box coordinates passed to it just to get a list of the available maps.   And then the downloaded data has a UTF-8 tag, so is refusing to parse it.

I have not started looking at this yet or at Thunderforest.com.   Thunderforest terms of use want a copyright notice to be displayed when D-Rats starts and is configured to use their maps.