ham-radio-software/D-Rats #80: Coredumps on 0.3.10 beta2

dranch, 2020-08-11:
I started an email thread on the Groups.IO mail list and Maurizio added a FAQ entry on what to do next if anyone is  experiencing a coredump.  Here is what I'm observing maybe once every week or so leaving D-rats 0.3.10Beta2 running 24/7:

STDOUT on a Centos 6.8 host:
--
08/11/2020 10:30:58 Transport : Got a block: DDT2+: 0:1:5 DL0NOT-H->CQCQCQ (2QRA DL0NOT WinMor/ALE Station - QTH Karben JO40jf QRV]...[55])
Chat      : Got chat frame: DDT2+: 0:1:5 DL0NOT-H->CQCQCQ (2QRA DL0NOT WinMor/ALE Station - QTH Karben JO40jf QRV]...[55])
08/11/2020 10:30:58 Sessionmgr: Received block 0:5 for session `chat'
08/11/2020 10:30:58 Transport : Got a block: DDT2+: 0:1:5 VE7CUS->CQCQCQ (1Online (D-RATS)...[16])
Chat      : Got chat frame: DDT2+: 0:1:5 VE7CUS->CQCQCQ (1Online (D-RATS)...[16])
08/11/2020 10:30:58 Sessionmgr: Received block 0:5 for session `chat'
08/11/2020 10:30:59 Transport : Got a block: DDT2+: 0:1:5 NS2B->CQCQCQ (2Online (D-RATS)...[16])
Chat      : Got chat frame: DDT2+: 0:1:5 NS2B->CQCQCQ (2Online (D-RATS)...[16])
08/11/2020 10:30:59 Sessionmgr: Received block 0:5 for session `chat'
08/11/2020 10:30:59 Transport : Got a block: DDT2+: 0:1:5 N5JLR->CQCQCQ (1Online (D-RATS)...[16])
Chat      : Got chat frame: DDT2+: 0:1:5 N5JLR->CQCQCQ (1Online (D-RATS)...[16])
08/11/2020 10:30:59 Sessionmgr: Received block 0:5 for session `chat'
08/11/2020 10:30:59 Transport : Got a block: DDT2+: 0:1:5 AE5JO->CQCQCQ (1Online (D-RATS)...[16])
Chat      : Got chat frame: DDT2+: 0:1:5 AE5JO->CQCQCQ (1Online (D-RATS)...[16])
08/11/2020 10:30:59 Sessionmgr: Received block 0:5 for session `chat'
08/11/2020 10:30:59 Transport : Got a block: DDT2+: 0:1:5 K5MPH->CQCQCQ (1Online (D-RATS)...[16])
Chat      : Got chat frame: DDT2+: 0:1:5 K5MPH->CQCQCQ (1Online (D-RATS)...[16])
08/11/2020 10:30:59 Sessionmgr: Received block 0:5 for session `chat'
08/11/2020 10:30:59 Transport : Got a block: DDT2+: 0:1:5 KI4KWR->CQCQCQ (1Online (D-RATS)...[16])
Chat      : Got chat frame: DDT2+: 0:1:5 KI4KWR->CQCQCQ (1Online (D-RATS)...[16])
08/11/2020 10:30:59 Sessionmgr: Received block 0:5 for session `chat'
08/11/2020 10:31:00 Transport : Got a block: DDT2+: 0:1:5 KS2ARR-C->CQCQCQ (1Online (D-RATS)...[16])
Chat      : Got chat frame: DDT2+: 0:1:5 KS2ARR-C->CQCQCQ (1Online (D-RATS)...[16])
08/11/2020 10:31:00 Sessionmgr: Received block 0:5 for session `chat'
08/11/2020 10:31:00 Transport : Got a block: DDT2+: 0:1:5 KM6TRZ->CQCQCQ (2Don Ladera Ranch CA DM13EN...[27])
**
Gdk:ERROR:gdkregion-generic.c:337:miSetExtents: assertion failed: (pExtents->y1 < pExtents->y2)
**
Gdk:ERROR:gdkregion-generic.c:1123:miUnionNonO: assertion failed: (pReg->numRects<=pReg->size)
Aborted (core dumped)
--

I do see log items like the following when starting and running D-rats but they don't seem to hurt anything:
--
/home/archive/D-rats/D-Rats-0.3.10-beta-2/d_rats/ui/main_events.py:112: Warning: g_object_set_qdata: assertion `G_IS_OBJECT (object)' failed
  icon, message = model.get(iter, 1, 3)

/home/archive/D-rats/D-Rats-0.3.10-beta-2/d_rats/mainapp.py:1303: GtkWarning: gdk_window_invalidate_rect_full: assertion `GDK_IS_WINDOW (window)' failed
  gtk.main() 

08/11/2020 11:04:00 Transport : Sending block: DDT2+: 0:1:5 KI6ZHD->CQCQCQ (1Online (D-RATS)...[16])                                                    
Exception in thread Thread-17:                                                                                                                          
Traceback (most recent call last):                                                                                                                      
  File "/usr/lib64/python2.6/threading.py", line 532, in __bootstrap_inner                                                                              
    self.run()                                                                                                                                          
  File "/usr/lib64/python2.6/threading.py", line 484, in run                                                                                            
    self.__target(*self.__args, **self.__kwargs)                                                                                                        
  File "/home/archive/D-rats/D-Rats-0.3.10-beta-2/d_rats/map_sources.py", line 161, in __thread_fn                                                      
    self.do_update()                                                                                                                                    
  File "/home/archive/D-rats/D-Rats-0.3.10-beta-2/d_rats/map_sources.py", line 244, in do_update                                                        
    printlog("[NBDC] Failed to fetch info for %i: %s" % (self.__buoy, e))                                                                               
TypeError: %d format: a number is required, not str   

08/11/2020 11:04:01 Transport : Sending block: DDT2+: 0:1:5 KI6ZHD->CQCQCQ (1Online (D-RATS)...[16])                 
Entity: line 516: parser error : Opening and ending tag mismatch: link line 71 and head                              
</head>                                                                                                              
       ^                                                                                                             
Entity: line 594: parser error : Entity 'nbsp' not defined                                                           
   <b><a href="https://water.usgs.gov"><u>USGS Water Resources</u></a></b>&nbsp;                                     
                                                                               ^                                     
Entity: line 594: parser error : Entity 'nbsp' not defined                                                           
a href="https://water.usgs.gov"><u>USGS Water Resources</u></a></b>&nbsp; &nbsp;                                     
                                                                               ^                                     
Entity: line 686: parser error : Opening and ending tag mismatch: input line 685 and span                            
      </span>                                                                                                        
             ^                                                                                                       
[... 68 lines cut ...]
#1  0x00000033afa33cd5 in ?? ()
#2  0x0000000000000001 in ?? ()
#3  0x00007fffbc1a4490 in ?? ()
#4  0x00007f2e14018760 in ?? ()
#5  0x00000033b1664871 in ?? ()
#6  0x00007f2e14018760 in ?? ()
#7  0x00000000b165fe31 in ?? ()
#8  0x0000000001eb1e48 in ?? ()
#9  0x00007f2e14018760 in ?? ()
#10 0x0000000001eb1e30 in ?? ()
#11 0x0000000001f88d00 in ?? ()
#12 0x0000000000000001 in ?? ()
#13 0x00007f2e14018760 in ?? ()
#14 0x0000000001eb1e30 in ?? ()
#15 0x0000000001f88d00 in ?? ()
#16 0x0000000000000001 in ?? ()
#17 0x0000000000000077 in ?? ()
#18 0x0000000000000001 in ?? ()
#19 0x00000033b1666e56 in ?? ()
#20 0x0000000000000060 in ?? ()
#21 0x0000003da8682a14 in ?? ()
#22 0x0000000000000020 in ?? ()
#23 0x0000000000000000 in ?? ()
(gdb) q
--

dranch, 2020-08-29:
Got another coredump last night after running for what seemed running reliably two weeks or so:

08/29/2020 05:41:12 Transport : Got a block: DDT2+: 0:1:5 KG5CEN-1->CQCQCQ (2Online (D-RATS)...[16])                                                                                                           
Chat      : Got chat frame: DDT2+: 0:1:5 KG5CEN-1->CQCQCQ (2Online (D-RATS)...[16])                                                                                                                            
08/29/2020 05:41:12 Sessionmgr: Received block 0:5 for session `chat'                                                                                                                                          
08/29/2020 05:41:12 Transport : Got a block: DDT2+: 0:1:5 AC0HY  X->CQCQCQ (1Online (D-RATS)...[16])                                                                                                           
*** glibc detected *** /usr/bin/python: double free or corruption (fasttop): 0x00007f32e000a5d0 ***                                                                                                            
Chat      : Got chat frame: DDT2+: 0:1:5 AC0HY  X->CQCQCQ (1Online (D-RATS)...[16])                                                                                                                            
08/29/2020 05:41:12 Sessionmgr: Received block 0:5 for session `chat'                                                                                                                                          
======= Backtrace: =========                                                                                                                                                                                   
/lib64/libc.so.6[0x33afa75e5e]                                                                                                                                                                                 
/lib64/libc.so.6[0x33afa78cad]                                                                                                                                                                                 
/usr/lib64/libgdk-x11-2.0.so.0(gdk_region_intersect+0x6f)[0x3da8634b1f]                                                                                                                                        
/usr/lib64/libgdk-x11-2.0.so.0[0x3da8642b7f]                                                                                                                                                                   
/usr/lib64/libgdk-x11-2.0.so.0(gdk_window_process_updates+0x174)[0x3da8647814]                                                                                                                                 
/usr/lib64/libgtk-x11-2.0.so.0[0x3da8c6a89f]                                                                                                                                                                   
/lib64/libgobject-2.0.so.0(g_closure_invoke+0x15e)[0x33b2a0e3de]                                                                                                                                               
/lib64/libgobject-2.0.so.0[0x33b2a248c5]                                                                                                                                                                       
/lib64/libgobject-2.0.so.0(g_signal_emit_valist+0x7e6)[0x33b2a25d76]                                                                                                                                           
/lib64/libgobject-2.0.so.0(g_signal_emit+0x83)[0x33b2a26333]                                                                                                                                                   
/usr/lib64/libgtk-x11-2.0.so.0(gtk_adjustment_value_changed+0x5a)[0x3da8a79eaa]                                                                                                                                
/usr/lib64/libgtk-x11-2.0.so.0[0x3da8c5d245]                                                                                                                                                                   
/usr/lib64/libgtk-x11-2.0.so.0[0x3da8c71602]                                                                                                                                                                   
/usr/lib64/libgtk-x11-2.0.so.0[0x3da8c71853]                                                                                                                                                                   
/usr/lib64/libgtk-x11-2.0.so.0[0x3da8c7191c]                                                                                                                                                                   
/usr/lib64/libgdk-x11-2.0.so.0[0x3da861ea96]                                                                                                                                                                   
/lib64/libglib-2.0.so.0(g_main_context_dispatch+0x1f5)[0x33b1640555]                                                                                                                                           
/lib64/libglib-2.0.so.0[0x33b1644d08]                                                                                                                                                                       [... cut ...]
7f33130b4000-7f33130bb000 r--s 00000000 fd:00 399854                     /usr/lib64/gconv/gconv-modules.cache
7f33130bb000-7f33130bc000 rw-p 00000000 00:00 0
7fffaf4d2000-7fffaf567000 rw-p 00000000 00:00 0                          [stack]
7fffaf5b9000-7fffaf5bb000 r--p 00000000 00:00 0                          [vvar]
7fffaf5bb000-7fffaf5bd000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted (core dumped)

wb8tyw, 2021-12-30:
The USGS maps can not be used until a new API is written to fetch them.
Until then any attempt by D-rats to access USGS maps will fail in various ways.

wb8tyw, 2023-04-03:
Closing as this appears to be for the Python2 version of D-Rats.

closed 2023-04-03 by wb8tyw