ham-radio-software/D-Rats #89: xmlParseMemory failed on start and selection on messages inbox

mtb2450, 2021-02-02:
**Synopsis**

Selecting "inbox" on Message tab results in xmlParseMemory() error.
Can not see any new messages.
Problem is repeatable.
Reinstalled 0.3.9 and failure persists.
Used to reliably work.

The same traceback occurs on initial DRATS start.

**Configuration**

D-RATS Version: 0.3.9
OS: Windows 10  Home 64-bit, build 19042.746
PC: Dell Inspiron i7 with 8GB memory

**Traceback**

 File "d_rats\ui\main_messages.pyo", line 1118, in <lambda>
 File "d_rats\ui\main_messages.pyo", line 714, in open_folder
 File "d_rats\ui\main_messages.pyo", line 701, in refresh
 File "d_rats\ui\main_messages.pyo", line 662, in _update_message_info
 File "d_rats\formgui.pyo", line 622, in __init__
 File "libxml2.pyo", line 1285, in parseMemory
 parserError: xmlParseMemory() failed

maurizioandreotti, 2021-02-02:
does the log says which file was being checked before the issue happened? 

my best guess is that somefile in the inbox got corrupted of is invalid-
can you check the content of your inbox n the filesystem and move all messages to a temp place and see if it behaves correctly?

mtb2450, 2021-02-02:
Maurizio,

Should have dumped the full log in the earlier message, sorry.  Interesting comment on beautifulsoup.

There were six messages in my inbox C:\Users\bell2\AppData\Roaming\D-RATS-EV\messages\Inbox
As you suggested I moved them aside, restarted DRats and no longer have the xmlParseMemory failure.  One by one I moved each message into the inbox, restarting DRats each time.  I found the offending message and have included its xml below.

Anything I can do for more info to you?

Mark N7GRB

==========================================
Log from the original failure as seen right after starting DRats:

Mainapp   : Enabled debug log for Win32 systems
Version : headers={'User-Agent': 'd-rats/0.3.9 '}
C:\Users\bell2\Documents\Ham Radio\D-RATS\library.zip\BeautifulSoup.py:114: UserWarning: You are using a very old release of Beautiful Soup, last updated in 2011. If you installed the 'beautifulsoup' package through pip, you should know the 'beautifulsoup' package name is about to be reclaimed by a more recent version of Beautiful Soup which is incompatible with this version.

This will happen at some point after January 1, 2021.

If you just started this project, this is easy to fix. Install the 'beautifulsoup4' package instead of 'beautifulsoup' and start using Beautiful Soup 4.

If this is an existing project that depends on Beautiful Soup 3, the project maintainer (potentially you) needs to start the process of migrating to Beautiful Soup 4. This should be a relatively easy part of the Python 3 migration.

Qst       : FeedParser not available
Qst       : QSTWeatherWU class retired
Config    : FILE: C:\Users\bell2\AppData\Roaming\D-RATS-EV\d-rats.config
Mainapp   : Loading locale `en'
Mainapp   : Locale dir is: C:\Users\bell2\Documents\Ham Radio\D-RATS\locale
---------------------------------------------------------------------------
Mainapp   :  D-RATS v0.3.9  starting at Tue Feb 02 12:43:44 2021
Mainapp   : Platform Win32Platform:

  base:       C:\Users\bell2\AppData\Roaming\D-RATS-EV
  source_dir: C:\Users\bell2\Documents\Ham Radio\D-RATS
  OS version: Windows 8->10 (6, 2, 9200, 2, '') 
---------------------------------------------------------------------------
Mainapp   : load position from config file
Mainapp   : Configuring the Static position: 34.018300,-84.040800
Mainapp   : Mapserver not active: False, call:; N7GRB
Gps       : StaticGPSPosition: UNKNOWN reporting 34.0183,-84.0408@1080 ft at 12:43:44
Mainapp   : create map window object-----
Mainapp   : create map window object: connect object-----
Mainapp   : query local gps device to see our current position
Gps       : ----------
Gps       : Failed to parse DPRS comment: BN  *20 
Gps       : CHECKSUM(N7GRB): 46 != 32
Gps       : Checksum : *20 
Gps       : _checksum: *2E 
Gps       : astidx   : 4 
Gps       : ----------
Mainapp   : load main window with self config
C:\Users\bell2\Documents\Ham Radio\D-RATS\library.zip\d_rats\mainwindow.py:286: GtkWarning: gtk_notebook_set_tab_label: assertion `GTK_IS_WIDGET (child)' failed
---- GUI Exception ----
Traceback (most recent call last):

  File "d_rats\ui\main_messages.pyo", line 1118, in <lambda>
  File "d_rats\ui\main_messages.pyo", line 714, in open_folder
  File "d_rats\ui\main_messages.pyo", line 701, in refresh
  File "d_rats\ui\main_message
s.pyo", line 662, in _update_message_info
  File "d_rats\formgui.pyo", line 622, in __init__
  File "libxml2.pyo", line 1285, in parseMemory
parserError: xmlParseMemory() failed

---- End ----

======================================================
Message causing failure
======================================================

{
  "ShowDR": "",
  "txtfiletoread": "",
  "inc_name": "GA ARES State Digital Net",
  "to_name": "KX4MAT / NCS",
  "txtStr": "",
  "fm_name": "Mark Bell, AEC Gwinnett Co.",
  "Subjectline": "GA ARES Digital Training Msg",
  "Mdate": "2020-10-04",
  "Mtime": "12:33",
  "Message": "\nPlease check in Mark, N7GRB\nAEC Public Health, Gwinnett Co., Metro District\n\nOperating on commercial power with available battery and generator backup.\n\n73's Mark\n",
  "Message2": "\nPlease check in Mark, N7GRB\nAEC Public Health, Gwinnett Co., Metro District\n\nOperating on commercial power with available battery and generator backup.\n\n73's Mark\n",
  "Approved_Name": "Mark / N7GRB",
  "Approved_PosTitle": "ARES Operator / AEC Gwinnett",
  "ResetForm": "Reset Form ",
  "testCall": "N7GRB",
  "parseme": "GA ARES State Digital Net\tKX4MAT / NCS\tMark Bell, AEC Gwinnett Co.\tGA ARES Digital Training Msg\t2019-10-27\t15:38\tPlease check-in N7GRB, Mark in Suwanee <br>Gwinnett ARES - Metro District<br>Operating on commercial power - emergency battery and generator available\tMark / N7GRB\tARES Operator / AEC Gwinnett\t\tRoutine\tfalse\t\t\n",
  "Templateversion": "ICS 213  v.41.8"
}

closed 2023-04-04 by wb8tyw