ham-radio-software/D-Rats #135: mapdraw error on raspberrypi

wb8tyw, 2022-08-27:
[maurizioandreotti](https://github.com/maurizioandreotti) commented [on Mar 22](https://github.com/wb8tyw/D-Rats/issues/120#issue-1176506821)

today, testing latest code on raspberrypi i cannot see any maps shown.

in the console i read:
```text
Traceback (most recent call last):
  File "/home/pi/drats2/D-Rats-python3_tyw/d_rats/map/mapdraw.py", line 179, in handler
    self.expose_map()
  File "/home/pi/drats2/D-Rats-python3_tyw/d_rats/map/mapdraw.py", line 393, in expose_map
    tile = center + (i - delta_w, j - delta_h)
TypeError: unsupported operand type(s) for +: 'NoneType' and 'tuple'
```

wb8tyw, 2022-08-27:
Can you reproduce this on demand?

What map url, center coordinates, and zoom level were in effect?

This looks like something in the map setup failed much earlier and there should have been log messages to that effect.

The default center location comes for the configuration for the user running d-rats. I suspect that nothing is checking to see if a valid value has been entered.

wb8tyw, 2023-04-03:
Is this still an issue?

maurizioandreotti, 2023-04-19:
can be closed

wb8tyw, 2023-04-22:
Leaving this open until we can verify that there is no path where "center" can be None at this point.   Something has to be wrong in the code.