ham-radio-software/D-Rats #143: Unused modules in D-Rats

wb8tyw, 2022-08-27:
Ticket for tracking unused modules in d-rats.

- wu.py - Was once possibly used in qst.py.
- reqobject.py - No references to it other than utils.py mentions it.
- callsigns.py
- ax25.py

maurizioandreotti, 2022-09-08:
> wu.py - Was once possibly used in qst.py.
yes this is possible, as at some point in time I had substituted the Weather Undergound part with what is there now

wb8tyw, 2023-07-08:
Unused class ProgressDialog found in miscwidgets.py.
Current best practices are generally one major class per module, so moving ProgressDialog to its own progressdialog.py module.  Then later we will see if there is someplace we can use this to improve the UI or reduce duplication.