ham-radio-software/D-Rats #149: D-Rats on new install creating download-dir before user confirms selection

wb8tyw, 2022-08-27:
When D-Rats starts up with no configuration, it attempts to create the download directory in a default location before the user has gone into the configuration and confirmed the directory to use for downloads.

There may be other issues similar to this.

On Windows 10, there is a report that trying to create the download directory in that default location is failing.

maurizioandreotti, 2022-09-13:
email thread about this issue: https://groups.io/g/d-rats/topic/93610916#2441
wiki page addressing how to install d-Rats on Windows 10 with onedrive: https://github.com/ham-radio-software/D-Rats/wiki/132---folder-problem-installing-D-Rats-on-Windows-10

wb8tyw, 2022-09-14:
Setting up a Windows 10 user with a Microsoft account (Default option) apparently enables Onedrive.   There is a lot of stuff that IDEs and browsers put in the user directories that can wipe out the free quota of OneDrive very quickly if they are not excluded.   We may need to provide a link on how to exempt directories from the one-drive.   The map tile cache is something that should not go to the one-drive.

closed 2023-05-13 by wb8tyw