ham-radio-software/D-Rats #152: lzhuf binary should not be packaged in python egg

wb8tyw, 2022-08-27:
Current python egg building script for d-rats is building and installing a script with a lzhuf binary.
This makes the resulting egg only installable on that specific platform.

The proper procedure is for a pip install of the resulting egg to do a local build of the lzhuf binary, which means that the end user getting the egg must already have to pre-installed the tools to build it.

In the future, we should look into getting the lzhuf utility pre-built in a distro specific package that can be pre-installed before installing d-rats.

yo2loj, 2022-08-30:
Would it make sense to do the effort to have a Python implementation of 
lzhuf?

wb8tyw, 2022-08-30:
If someone has the time and the knowledge to do so, that could make it easier.
I would still prefer it to be in a separate package from d-rats.

In either case, this ham-radio-software can host the repository for it.

closed 2022-12-03 by wb8tyw