ham-radio-software/D-Rats #154: Deprecate use of RAR format for distributions

wb8tyw, 2022-08-27:
These are the compression tools universally used for distributions.

ZIP is built into Microsoft Windows and is easily available from a trusted source for all distributions.

TAR with GZIP is the next most popular as it is supported by the PIP install program.

TAR with XZ or BZIP2 is also an option that should be supported by PIP install programs.

Use ZIP if targeting Microsoft Windows, and TAR with compression for PIP and all other platforms.

The use of the RAR protocol sends the users scrambling to find a program to unpack it, and many of the alleged providers that show up in internet searches are fake install sites. The fake install sites will generally install malware on the target system that is a real pain to clean out, and in many cases will have modified the desired program to include the malware.

dslotter, 2022-08-27:
I support this reasoning. Zip and Tarballs are well-supported formats.

maurizioandreotti, 2022-09-08:
I am not sure if it's worth spending more time on this topic, as I also agree.  

If i am not mistaken, the latest compile of D-Rats for ms-windows was already in zip package. (I never got to compile it for other platforms, and source code was usually downloadable from github in the supported formats ). 

I dont remember what was the reason for past use of rar format, i guess I needed it for some problems with zip.

wb8tyw, 2022-09-09:
RAR compression is copyrighted and not open source according the references that I have seen.
I can not find a free to use legal copy of RAR compression tool with quick search.

The existing Python 2 packaging procedure of creating a an embedded python interpreter which appears to be what was in the zip file does not appear to be an option for Python 3.

So currently our short term option is a pip installable tarball, (There is a work around for using the wheel tarball for Mac-OS).

The pip installable tarball should not include any binaries like lzhuf.

So step 1 is to get lzhuf as its own native package for all the platforms that we care about.
Step 2 is to find a way for non-computer literate people to install a python environment, D-Rats and an optional lzhuf package.

dslotter, 2022-10-11:
Is it acceptable from a license perspective to have a master installer for
D-Rats which also installs a Python runtime along with an optional lzhuf
package? If the answer to that is yes, I would say the technical
implementation is just a matter of effort.

wb8tyw, 2022-10-11:
https://github.com/ham-radio-software/D-Rats/issues/159 covers that issue.

Since writing this ticket, I have found a supported tool to create d-rats bundled with a python3 runtime.

closed 2022-12-03 by wb8tyw