ham-radio-software/D-Rats #159: Setup script for pyinstaller

wb8tyw, 2022-09-09:
Need to set up the following:

- Powershell script to install python on Windows from official python.org repository.
- Powershell script to create a virtualenv for pip installs.
- Fresh clone of a d-rats commit into a work directory.
- Pip install of the requirements for d-rats python environment
- pip install of PyInstaller
- build a python tarball 
- pip install the D-Rats
- run the pyinstaller to create the windows exe
- Verify that the resulting issue works.
- Add a github action run this script on pull requests and on the master branch merges.

wb8tyw, 2022-09-10:
https://pyinstaller.org/en/stable/

wb8tyw, 2022-10-11:
Current plans are to see if I can use the lzhuf from a prebuilt MSI before using the pyinstaller when building an MSI package with the exe file from pyinstaller file, so there is one MSI file to install.

wb8tyw, 2022-11-07:
Unfortunately I have hit blocker on doing this.

The current python PyPi modules for GTK+ that are required for D-Rats need to build some binaries with Visual Studio.   That build is failing, and the internet searches so far have not show any solution.

All threads that I have found so far have resulted in the person giving up on a native Python.org  Windows solution and installing msys2 instead.

Leaving this ticket open in case someone else has better luck.

SpudGunMan, 2022-11-07:
It's been a second since I have ran windows. But what about a different path? Make a docker image for windows to hold the depends. Make just the minimal docker instance to allow execution and call it a day?

wb8tyw, 2022-11-08:
If someone wants to provide in Wiki articles how to build that docker image and how to install it, I am sure that no one would object.

Like Microsoft WSL, the Windows docker system only runs on some Windows 10 platforms.  Current documentation I just checked shows that it requires Microsoft WSL 2nd Generation.

Unlike Microsoft WSL, the Docker Desktop that may be required, is only free for a subset of users.   As such, I will not be installing it on any of my systems that may otherwise be able to run it.

If WSL is installed, the Ubuntu image can be installed and may be simpler just to install D-Rats and optionally LZHUF on it instead of maintaining a Docker image.
That is a totally free solution, but not high on my personal testing agenda right now.  If someone wants to add a Wiki page on how to do this, it would be appreciated.

Currently only Cygwin and Msys2 are the only two solutions I know of with totally free licenses that will run on most Microsoft Windows systems currently in use.   It is unknown how long they will be available for install on Microsoft Windows that is older than Windows 10.

It would be nice if someone could create a wiki page on how to install msys2 and then install d-rats and lzhuf.

Currently a fixed lzhuf package is available from the files section of the D-Rats mailing list server.   Currently the D-Rats python tarball does not use it, but in the near future, the D-Rats python tarball will need it installed or Winlink support will be disabled.,

SpudGunMan, 2022-11-08:
Also this [image](https://github.com/docker-library/python) is likely better.

use the template files copy the project into docker add the pip (compiled a list on my scratch decker file last post) 
and finally 
`CMD cd /D-rats && [ "python", "d-rats.py" ]`

wb8tyw, 2023-03-30:
Found some more leads to investigate for creating a standalone Msyss2 directory for Microsoft Windows.

https://www.msys2.org/wiki/Distributing/

Another option would be to come up with a package that runs a script that installs Msys2 mingw64 if it is not present and also installs D-Rats on it, which would take less work for the end user.   I am thinking that this may be the best option to look at.

It also turns out that the current Windows Subsystem for Linux does not have access to USB/Serial p;orts.   There is an experimental hack for enabling that support.  However the public documentation for Windows Subsystem for Linux is lagging the actual product, and my last install had problems because some of the public instructions were wrong, and that was just installing the supported features, not the experimental hacks.