ham-radio-software/D-Rats #162: python package building broken

wb8tyw, 2022-11-11:
Not sure if it something in my local environment or unexpected change in a future PR.

On my main developement system, I have not tried the python -m build procedure since I upgraded it from 20 (buster) to 21 (bullseye).

When I tried to run it with just a change to not include the LZHUF binary that is now in a separate Debian package, it fails to run with a trace back, the last line has ModuleNotFoundErrror for d_rats.

Since we are not in compliance with the strongly recommended setup for a python project for the new packaging, it is unlikely that I can expect much help from the Python community in chasing this down.

On Microsoft Windows 7, msys2 the python -m build has a number of tracebacks in it for 'ValueErrror: The underlying buffer has been detached'.   It still claims to have built the python tarball though.

With Microsoft Windows 7, we have the additional issue that the Python Upstream is in the process of dropping support for Windows 7, and will likely be dropping support for anything older than Windows 10 at the end of this year.

An Internet search indicates either that this issue I am seeing was fixed a few years earlier, or is caused by something in attempting to use the Microsoft Visual C++ that was installed to build lzhuf, that may not occur on newer versions of Microsoft Windows.

wb8tyw, 2022-11-11:
Renaming the setup.py that is no longer supported to be used "fixes" the issue on Anti-X linux.
It does not change the behavior on Microsoft Windows 7.

Unfortunately right now, removing the setup.py file means that all the stuff in it that we currently need doing is not getting done, so that requires more work.

SpudGunMan, 2022-11-11:
I’m looking at this from far away and little context to the project. But looking at setup.py a lot of bash/batch work environment set up and the build of the lzhuf in question. Any reason you can’t convert this to a .msi package and forgo make 

https://github.com/ham-radio-software/lzhuf

wb8tyw, 2022-11-11:
I am confused about your comment.  lzhuf is already buildable as an MSI package and as a number of Ubuntu/Debian packages.  The pre-built packages are available or download by members of the D-Rats groups.io mailing list.

There is a https://github.com/ham-radio-software/D-Rats/pull/161 in progress, that will be followed by another Pull request that will have D-Rats use lzhuf from that project, and remove it from D-Rats.  Until that is done, D-Rats will not know how to find that.  While preparing the second PR, as part of the testing, I did a test of the package building.

For D-Rats, we need to first build it as a Python tarball package, and the Python standards have evolved to the point that the previous methods have been deprecated.  But the new supported
method does not deal with building the message catalogs needed for non-English locales, and also may not do the "towncrier build".  And the new supported method reallly wants other significant changes how the D-Rats files are in their directory.   There is another ticket about that.

SpudGunMan, 2022-11-11:
as mentioned I was too far away to toss in a comment. understand.

wb8tyw, 2022-11-11:
I do not want to discourage you, we need all the help on this project that we can get.

The current big blocker is not having easy ways for non-programmers to just install and run it.

SpudGunMan, 2022-11-11:
You didn’t! Yup, not being a windows user slows me down. I will keep lurking for now.

wb8tyw, 2022-11-11:
By providing a lzhuf MSI file and if we document the Msys2 install and use, essentially there should be little difference in supporting Windows and Unix like platforms.

So this python packaging issue is a generic python issue.

What it appears is that none of the easy to configure or python backends can support running msgfmt, so we will have to put a wrapper around the python -m build command to build them first, and control our own destiny.

With lzhuf now a separate install, we will be able to build pip installable D-Rats package for any platform on any platform.   It also means only one tarball to install what ever platform people are running.

closed 2022-12-03 by wb8tyw