ham-radio-software/D-Rats #170: Crash when creating new message

Eric-Sparks, 2023-03-29:
On the message tab, when I select the new message button (not the drop down) I get a crash:

```
Traceback (most recent call last):

  File "/home/echristensen/.local/lib/python3.11/site-packages/d_rats/ui/main_messages.py", line 128, in _new_msg
    dialog = inputdialog.ChoiceDialog(forms.keys(),
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/home/echristensen/.local/lib/python3.11/site-packages/d_rats/inputdialog.py", line 102, in __init__
    default = choices[0]
              ~~~~~~~^^^

TypeError: 'dict_keys' object is not subscriptable
```

wb8tyw, 2023-03-30:
Since you have d-rats installed in a path different than what is on any of my test machines, I would be useful to know:

- What exact d-rats version?
- What is the platform it is installed on?
- What was the installation command?

wb8tyw, 2023-04-03:
Possible duplicate of https://github.com/ham-radio-software/D-Rats/issues/118

wb8tyw, 2023-04-11:
line 128 of d_rats/ui/main_messages.py needs to be changed to:
`            dialog = inputdialog.ChoiceDialog(list(forms),
`
Now looking to see if that fix is needed anywhere else.

closed 2023-05-13 by wb8tyw