ham-radio-software/D-Rats #173: Windows install - option 2 -

maurizioandreotti, 2023-04-06:
Following instructions:
- https://github.com/ham-radio-software/D-Rats/wiki/010.010-Installation-of-D-Rats-on-Microsoft-Windows
- Option 2, Installing the latest Python "tarball" for D-Rats

After having downloaded the tar file, when I issue the `ls $USEROPROFILE\\downloads\\ `command it ends with permission error:

```
$ ls $USEROPROFILE\\downloads\\
ls: cannot access '\downloads\': No such file or directory
(d-rats_venv)
mauri@T450-Maurizio MINGW64 ~

```

maurizioandreotti, 2023-04-06:
moved file to 
`C:\msys64\home\mauri\`
in order to proceed with install
` pip install D-Rats-0.4.0.tar.gz
`

wb8tyw, 2023-04-06:
Fixed typo on [Installation of D-Rats on Microsoft Windows](https://github.com/ham-radio-software/D-Rats/wiki/010.010-Installation-of-D-Rats-on-Microsoft-Windows)