ham-radio-software/D-Rats #174: drats 4 - no maps download

maurizioandreotti, 2023-04-06:
using D-rats 4, installed as "option 2"

while it uses the tiles already downloaded, moving to a new location, it doesnt donwload the additional tiles.

I get this error in the pop-up window:

Traceback (most recent call last):
```

  File "C:/msys64/home/mauri/d-rats_venv/lib/python3.10/site-packages/d_rats/map/mapdraw.py", line 179, in handler
    self.expose_map()

  File "C:/msys64/home/mauri/d-rats_venv/lib/python3.10/site-packages/d_rats/map/mapdraw.py", line 404, in expose_map
    self.draw_tile(None,

  File "C:/msys64/home/mauri/d-rats_venv/lib/python3.10/site-packages/d_rats/map/mapdraw.py", line 372, in draw_tile
    surface = self.broken_tile()

  File "C:/msys64/home/mauri/d-rats_venv/lib/python3.10/site-packages/d_rats/map/mapdraw.py", line 202, in broken_tile
    self.__broken_tile = cairo.ImageSurface.create_from_png(broken_path)

cairo.Error: file not found

```
and last lines in the console are

[](04/06/2023 04:16:21:INFO:MainApp:_static_gps: Mapserver active: True ca;; IZ2LXI-W
04/06/2023 04:16:21:INFO:MainApp:preparing our gps fix in JSON :b'{ "lat": "45.800456", "lng": "9.428377", "station": "IZ2LXI-W"
, "comments": "altitude: 0","timestamp": "2023-04-06 02:16:21"  }'
04/06/2023 04:16:21:INFO:MainApp:Socket Created
04/06/2023 04:16:21:INFO:MainApp:Connecting to: localhost:9500
04/06/2023 04:16:22:INFO:SocketDataPath:getline: Host does not require authentication
04/06/2023 04:16:23:INFO:MainApp:Failed to create socket.
Traceback (most recent call last):
  File "C:/msys64/home/mauri/d-rats_venv/lib/python3.10/site-packages/d_rats/mainapp.py", line 433, in callback_gps
    sock.connect((mapserver_ip, mapserver_port))
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it
04/06/2023 04:16:23:INFO:StaticGPSSource:StaticGPSPosition: UNKNOWN reporting 45.8005,9.4284@0 m at 04:16:23
04/06/2023 04:16:23:INFO:MainApp:_refresh_map: reconfigure Mapwindow with new map
04/06/2023 04:16:23:INFO:MainApp:_refresh_map: reconfigured mapurl to: http://a.tile.opentopomap.org/
04/06/2023 04:16:23:INFO:MainApp:_refresh_map: reconfigured mapkey to:
04/06/2023 04:16:29:INFO:MainApp:Showing Map Window
04/06/2023 04:16:34:INFO:D-Rats:---- GUI Exception ----
%s
---- End ----
Stack (most recent call last):
  File "C:\msys64\home\mauri\d-rats_venv\bin\d-rats.py", line 240, in <module>
    main()
  File "C:\msys64\home\mauri\d-rats_venv\bin\d-rats.py", line 234, in main
    app.main()
  File "C:/msys64/home/mauri/d-rats_venv/lib/python3.10/site-packages/d_rats/mainapp.py", line 2124, in main
    self.run(None)
  File "C:/msys64/mingw64/lib/python3.10/site-packages/gi/overrides/Gio.py", line 42, in run
    return Gio.Application.run(self, *args, **kwargs)
  File "C:\msys64\home\mauri\d-rats_venv\bin\d-rats.py", line 93, in handle_exception
    MODULE_LOGGER.info("---- GUI Exception ----\n%s\n---- End ----\n",

)

wb8tyw, 2023-04-07:
Looks like mapdraw.py is not using the correct method to find the broken tile PNG file for the PIP installable package.

maurizioandreotti, 2023-04-18:
I can confirm that map tile download now works.

closed 2023-05-13 by wb8tyw