ham-radio-software/D-Rats #175: drats 4 - error at start up opening drats 4 with former config file

maurizioandreotti, 2023-04-06:
Having done a fresh install of d-rats option 2 on a Win-10 pc where  it was installed 0.3.9 I get this error:
, which can be bypased by pressing "ignore all" and after all seems working

from Info window/ Debug: Traceback (most recent call last):

 ```
 File "C:/msys64/home/mauri/d-rats_venv/lib/python3.10/site-packages/d_rats/formgui.py", line 1076, in __init__
    self.doc = etree.parse(self._filename)

  File "src/lxml/etree.pyx", line 3541, in lxml.etree.parse

  File "src/lxml/parser.pxi", line 1879, in lxml.etree._parseDocument

  File "src/lxml/parser.pxi", line 1905, in lxml.etree._parseDocumentFromURL

  File "src/lxml/parser.pxi", line 1808, in lxml.etree._parseDocFromFile

  File "src/lxml/parser.pxi", line 1180, in lxml.etree._BaseParser._parseDocFromFile

  File "src/lxml/parser.pxi", line 618, in lxml.etree._ParserContext._handleParseResultDoc

  File "src/lxml/parser.pxi", line 728, in lxml.etree._handleParseResult

  File "src/lxml/parser.pxi", line 657, in lxml.etree._raiseParseError

  File "C:/Users/mauri/AppData/Roaming/D-RATS-EV/messages/Inbox/form_06252022_090631.xml", line 1

lxml.etree.XMLSyntaxError: Start tag expected, '<' not found, line 1, column 1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "C:/msys64/home/mauri/d-rats_venv/lib/python3.10/site-packages/d_rats/ui/main_messages.py", line 102, in <lambda>
    lambda x, y: self._messages.open_folder(y))

  File "C:/msys64/home/mauri/d-rats_venv/lib/python3.10/site-p

```

from console:

`
$ d-rats.py
04/06/2023 03:28:38:INFO:D-Rats:main: re-config option found -- Reconfigure D-rats
04/06/2023 03:28:38:INFO:D-Rats:main: args.config = C:/Users/mauri/AppData/Roaming/D-RATS-EV
04/06/2023 03:28:38:INFO:Mainapp:Enabled debug log for Win32 systems
04/06/2023 03:28:39:INFO:DratsConfig:File C:/Users/mauri/AppData/Roaming/D-RATS-EV/d-rats.config
04/06/2023 03:28:39:INFO:MainApp:_refresh_lang: Loading locale `it'
04/06/2023 03:28:39:INFO:MainApp:_refresh_lang: Locale dir is: C:/msys64/home/mauri/d-rats_venv/shar
e/d-rats/locale
04/06/2023 03:28:39:ERROR:MainApp:_refresh_lang: Messages catalog file missing  for it.  Need to use
 'msgfmt tool to generate.
04/06/2023 03:28:39:INFO:MainApp:D-RATS v0.4.0 starting at Thu Apr  6 03:28:39 2023 on Platform Win3
2Platform:
  configuration:       C:/Users/mauri/AppData/Roaming/D-RATS-EV
  system_data: C:/msys64/home/mauri/d-rats_venv/share/d-rats
  OS version: Windows 10 (10, 0, 19044, 2, '')
04/06/2023 03:28:39:INFO:MainApp:load position from config file
04/06/2023 03:28:39:INFO:MainApp:_static_gps: Configuring the Static position: 45.800456,9.428377
04/06/2023 03:28:39:INFO:MainApp:_static_gps: Mapserver active: True ca;; IZ2LXI-W
04/06/2023 03:28:39:INFO:MainApp:preparing our gps fix in JSON :b'{ "lat": "45.800456", "lng": "9.42
8377", "station": "IZ2LXI-W", "comments": "altitude: 0","timestamp": "2023-04-06 01:28:39"  }'
04/06/2023 03:28:39:INFO:MainApp:Socket Created
04/06/2023 03:28:39:INFO:MainApp:Connecting to: localhost:9500
04/06/2023 03:28:41:INFO:MainApp:Failed to create socket.
Traceback (most recent call last):
  File "C:/msys64/home/mauri/d-rats_venv/lib/python3.10/site-packages/d_rats/mainapp.py", line 433,
in callback_gps
    sock.connect((mapserver_ip, mapserver_port))
[... 217 lines cut ...]
ports_38 = False,net:wa7dre.ratflector.com:9000,,True,True,XLX - wa7dre

[incoming_email]
incoming_email_0 = pop.gmail.com,maurizio.iz2lxi@gmail.com,Helena.1973,6,True,995,Form,True
incoming_email_1 = pop.mail.yahoo.com,iz2lxi@yahoo.it,helena1973,5,True,995,Form,False

[email_access]

[quick]
20220228212748 = Ciao all

[tcp_out]

[buoys]

[qst_20220316195246]
enabled = False
freq = 60
[d-rats .config.txt](https://github.com/ham-radio-software/D-Rats/files/11165614/d-rats.config.txt)

content = Current/Lecco
type = OpenWeather
port = Current

`

wb8tyw, 2023-04-06:
The issue here is that D-Rats is checking to see if a form sent to you in your inbox is a valid form, and the form failed the check.

`C:/Users/mauri/AppData/Roaming/D-RATS-EV/messages/Inbox/form_06252022_090631.xml`

It would help to have a copy of the first line of this file, if it does not contain any sensitive information so that this bug can be reproduced on demand.  In this case, it looks like I just need to put a file with a similar name in the inbox that does contain valid XML.

The next thing is what should D-Rats do when this happens?  As this is in the inbox, it is something that has apparently been sent.  We do not know how this invalid xml file arrived, or if it got corrupted post reception.

We probably need to flag this file in the Inbox in some way to indicate that even though it claims to be an XML file, it is not an XML file. so the end user can deterine what they want to do with it.

maurizioandreotti, 2023-04-06:
> Xml form attached - it contains some sort  of binary, I guess it arrived
> corrupted (creation date on filesystem is 25th of June 2022,... I
> don't recall what it was supposed to be).
>
>
> Il giorno gio 6 apr 2023 alle ore 15:39 John E. Malmberg <

renamed as Zip, got more understandable,... renamed again as xlsx file...
it shows a table of data I did

I guess I used this file to do some upload/file transfer test,... but how
it ended up being in that folder with xml extension, I dont really know