ham-radio-software/D-Rats #187: D-Rats-0.4.1_finalize does not open on mac (version.py error)

maurizioandreotti, 2023-05-14:
Launching the D-Rats-0.4.1_finalize on my mac i get this error in console:

```
./d-rats.py                
Traceback (most recent call last):
 File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-0.4.1_finalize/./d-rats.py", line 37, in <module>
   from d_rats.version import __version__
 File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-0.4.1_finalize/d_rats/version.py", line 401, in <module>
   DRATS_VERSION_NUM = GLOBAL_VERSION.short_version
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-0.4.1_finalize/d_rats/version.py", line 111, in short_version
   self._get_short_version()
 File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-0.4.1_finalize/d_rats/version.py", line 101, in _get_short_version
   (cls._version['major'], cls._version['minor'],
    ~~~~~~~~~~~~^^^^^^^^^
KeyError: 'major'
```

I have not yet tested on the Windows PC...

wb8tyw, 2023-05-15:
Please show the output of "git describe --tags --dirty" where you are running d-rats.

I am expecting it to report an error.

Try running with log level set to debug for more information.

I think I see where the problem is.  If this is not a git checkout (From tarball), and PKG-INFO is missing, as would be from a PR tarball, the internal "_version" structure is not fully populated.

Will not affect release code tarballs or running from git clone directory, but still a bug that needs to be fixed.

maurizioandreotti, 2023-05-15:
bingo! 
the PKG-INFO file was completley missing in the downloaded code from https://github.com/ham-radio-software/D-Rats/archive/refs/heads/master.zip

copying the file from the D-Rats bugfix made it start. 

(for time being I have not installed git on the mac)

maurizioandreotti, 2023-05-15:
I am renaming the issue as it is not mac or wi, but zip file related

wb8tyw, 2023-05-24:
PKG-INFO is not supposed to be in master or the generated tarballs from D-Rats as per previous issues.

The version.py code needs to be able to work with out PKG-INFO being present.

closed 2023-05-30 by wb8tyw