ham-radio-software/D-Rats #190: Configuration sound file selection titled save file as

wb8tyw, 2023-05-31:
Clicking on the three dots for selecting a sound file to read in the sound configuration brings up a selection window titled "Save file as".

maurizioandreotti, 2023-06-13:
for me this can be closed, as I see the window "select a file to open" and button is "open",

closed 2023-06-21 by wb8tyw

flyingfrawg, 2023-06-21:
John and Maurizio,

I may have missed something about this previously:

For the current "0.4.1.dev8-g536233c" Messages and Chat tabs I see: 
"06/21/2023 13:21:38:INFO:DratsConfig:Error in config file: settings/msg_flush data 30.000000 is not an int"
pop up repeatedly in my MSYS2 MINGW64 shell status window.
I don't seem to be able to use Preferences>Messages>Queue flush interval to correct it.  The settings menu shows just "30". This issue does not appear to impact any Message or chat Chat function that I can tell. It appears to be an issue of the declared type for the value.

Is this of concern?

Jack - KD4IZ

maurizioandreotti, 2023-06-21:
Hi Jack
I think it's a conversion bug, as now - after setting the

   Preferences->messages->Queue flush interval

value to 31, i also get that  message:

06/21/2023 22:49:58:INFO:DratsConfig:Error in config file:
settings/msg_flush data 30.000000 is not an int

06/21/2023 22:50:28:INFO:DratsConfig:Error in config file:
settings/msg_flush data 30.000000 is not an int

Do you want to open an issue in the git hub issue list?

(or i do it :-)

thanks

maurizio

flyingfrawg, 2023-06-21:
Maurizio,I would be very grateful if you could do it. I am not very adept at filling bug reports and I am out the door to an emergency vehicle operating class in minutes also will be “out of pocket” tomorrow.Grazi!Jack SpitznagelSent from my iPhone