ham-radio-software/D-Rats #195: Macbook - "test sound" result in error

maurizioandreotti, 2023-06-05:
in sound config panel, having selected a sound, playing "test sound" results in this error in the pop-up :
```

Traceback (most recent call last):

  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-master/d_rats/config.py", line 922, in test_sound
    platform_info.play_sound(self.value)

  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-master/d_rats/dplatform_generic.py", line 443, in play_sound
    sound = AudioSegment.from_wav(soundfile)
            ^^^^^^^^^^^^

NameError: name 'AudioSegment' is not defined
```

while in the console it shows:

```
06/05/2023 18:03:32:INFO:DratsConfigWidget:Testing playback of /System/Library/Sounds/Morse.aiff
06/05/2023 18:03:32:INFO:MacOSXPlatform:play_sound: bydub and pyaudio not installed!
06/05/2023 18:03:32:INFO:D-Rats:---- GUI Exception ----
%s
---- End ----
Stack (most recent call last):
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-master/./d-rats.py", line 240, in <module>
    main()
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-master/./d-rats.py", line 234, in main
    app.main()
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-master/d_rats/mainapp.py", line 2124, in main
    self.run(None)
  File "/Users/maurizio/Virtualenvs/d-rats4/lib/python3.11/site-packages/gi/overrides/Gio.py", line 42, in run
    return Gio.Application.run(self, *args, **kwargs)
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-master/d_rats/mainwindow.py", line 244, in _activate_prefs
    saved = self._config.show(parent=self._window)
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-master/d_rats/config.py", line 2699, in show
    result = drats_ui.run()
  File "/Users/maurizio/Virtualenvs/d-rats4/lib/python3.11/site-packages/gi/overrides/Gtk.py", line 585, in run
    return Gtk.Dialog.run(self, *args, **kwargs)
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-master/./d-rats.py", line 93, in handle_exception
    MODULE_LOGGER.info("---- GUI Exception ----\n%s\n---- End ----\n",
```

installing bydub and pyaudio.... both are not in the scope of the "brew" package. 

hitting `pip3 install  bydub `--> no package with this name:

`ERROR: Could not find a version that satisfies the requirement bydub (from versions: none)
ERROR: No matching distribution found for bydub`

hitting `pip3 install pyaudio` results with a compilation error:
```
      src/pyaudio/device_api.c:9:10: fatal error: 'portaudio.h' file not found
      #include "portaudio.h"
               ^~~~~~~~~~~~~
      1 error generated.
      error: command '/usr/bin/clang' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pyaudio
Failed to build pyaudio
ERROR: Could not build whe

```

maurizioandreotti, 2023-06-05:
on mac installed pyaudio via:
```
brew remove portaudio
brew install portaudio
pip3 install pyaudio
```

this results with pyaudio installed in
`pyaudio in /Users/maurizio/Virtualenvs/d-rats4/lib/python3.11/site-packages (0.2.13)
`

wb8tyw, 2023-06-05:
pyaudo and pydub use "dynamic loading" to install the packages that is needed on the local system.  They apparently do not install any libraries that may be missing.

The play sound code is not supposed even try using the code if either pydub or pyaudo python packages are not installed.

I landed that pull request after Rod Whitaker verified that he could create sound on his Mac.

The bydub is a typo that was missed somewhere.  It should be pydub.

The error message about ffprobe being missing came up with a lot of search results including https://stackoverflow.com/questions/57350259/filenotfounderror-errno-2-no-such-file-or-directory-ffprobe-ffprobe

In general using a pip installed modules is to be done as a last resort if there is no native package for from the operating system or python vendor to supply that module.  Pip installed modules may not be fully tested on all platforms.

wb8tyw, 2023-06-05:
Looks like we have the a lot of issues in how to get a D-Rats compatible python environment on Mac OS that need to be worked out.

wb8tyw, 2023-06-11:
Reopened automatically closed bug because we still need Mac OS documentation to how to make test_sound work.

maurizioandreotti, 2023-06-13:
I installed ffmpeg via brew and now I get the following error in the pop-up when i try to play a sound

`Traceback (most recent call last):

  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-gh/d_rats/config.py", line 923, in test_sound
    platform_info.play_sound(self.value)

  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-gh/d_rats/dplatform_generic.py", line 442, in play_sound
    sound = AudioSegment.from_wav(soundfile)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/Users/maurizio/Virtualenvs/d-rats4/lib/python3.11/site-packages/pydub/audio_segment.py", line 808, in from_wav
    return cls.from_file(file, 'wav', parameters=parameters)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/Users/maurizio/Virtualenvs/d-rats4/lib/python3.11/site-packages/pydub/audio_segment.py", line 773, in from_file
    raise CouldntDecodeError(

pydub.exceptions.CouldntDecodeError: Decoding failed. ffmpeg returned error code: 1

Output from ffmpeg/avlib:

ffmpeg version 6.0 Copyright (c) 2000-2023 the FFmpeg developers
  built with Apple clang version 14.0.3 (clang-1403.0.22.14.1)
  configuration: --prefix=/opt/homebrew/Cellar/ffmpeg/6.0 --enable-shared --enable-pthreads --enable-version3 --cc=clang --host-cflags= --host-ldflags= --enable-ffplay --enable-gnutls --enable-gpl --enable-libaom --enable-libaribb24 --enable-libbluray --enable-libdav1d --enable-libmp3lame --enable-libopus --enable-librav1e --enable-librist --enable-librubberband --enable-libsnappy --enable-libsrt --enable-libsvtav1 --enable-libtesseract --enable-libtheora --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libxvid --enable-lzma --enable-libfontconfig --enable-libfreetype --enable-frei0r --enable-libass --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libspeex --enable-libsoxr --enable-libzmq --enable-libzimg --disable-libjack --disable-indev=jack --enable-videotoolbox --enable-audiotoolbox --enable-neon
  libavutil      58.  2.100 / 58.  2.100
  libavcodec     60.  3.100 / 60.  3.100
  libavformat    60.  3.100 / 60.  3.100
  libavdevice    60.  1.100 / 60.  1.100
  libavfilter     9.  3.100 /  9.  3.100
  libswscale      7.  1.100 /  7.  1.100
  libswresample   4. 10.100 /  4. 10.100
  libpostproc    57.  1.100 / 57.  1.100
[wav @ 0x15a804080] invalid start code FORM in RIFF header
/Users/maurizio/Downloads/Blow.aiff: Invalid data found when processing input`

maurizioandreotti, 2023-06-13:
....using a wav sound instead of. a system .aiff  D-rats on macos plays sounds - so, it works!