ham-radio-software/D-Rats #207: D-Rats -> File -> Tools -> Network Proxy (not found)

maurizioandreotti, 2023-06-20:
Calling menu:

**D-Rats -> File -> Tools -> Network Proxy **

On MacOS I get this error:
06/20/2023 09:59:07:INFO:MainWindow:activate_proxy: Running proxy: ['/Users/maurizio/Virtualenvs/d-rats4/bin/python3', 'd-rats_repeater']
/Users/maurizio/Virtualenvs/d-rats4/bin/python3: can't open file '/Users/maurizio/Virtualenvs/d-rats4/D-Rats/d-rats_repeater': [Errno 2] No such file or directory

maurizioandreotti, 2023-06-20:
To be honest I am not sure why this was implemented in this way as the repeater should be run by itself (with its params and console loggin etc .

Personally I would remove this link

wb8tyw, 2023-06-20:
It did not appear to be working for the python2 version, so I did not do anything with it for the python3 version.
I probably should have created a ticket.

It is one of those applications that needs to be launched via the GIO calls that do not work yet on Mac OS.

Based on the error message, it appears that it was expecting a symlink to the actual d-rats repeater or a script to launch it to be present at that directory, and symlinks do not work well in a cross platform repository, and I do not seen any scripts.

The name "proxy" indicates that it is intended as a buffer between the client and the network, but still not sure of the intent of this feature.

We may want to ask on the d-rats.groups.io mailing list if anyone cares if we just remove that menu item.

To properly implement the feature, d-rats needs to test that see that the proxy is configured as a radio so that it knows the port to check, and check if the port is in use.   Note that is separate from having it client actually having the radio enabled for the proxy.

maurizioandreotti, 2023-06-20:
> The name "proxy" indicates that it is intended as a buffer between the
client and the network, but still not sure of the intent of this feature.

From what I see, in 0.3.9 (on win10) that menu voice just open the local
ratflector:

[image: image.png]

the end result it's identical as opening  d-rats_repeater.exe with a double
click (and the configfile used by the raflector is the same in both cases).
in fact closing d-rats doesn't close the ratflector program, which remain
operational

launching d-rats_repeater from command line allows specify a different
config, etc etc

wb8tyw, 2023-06-20:
Something extra was added to the windows packaging of 0.3.9. to make it work.

This may also have been done for .deb or .rpm packaging of D-Rats.  The code to create those packages also never made it completely into the D-Rats official sourc.e.

The feature does not work from just using a copy of the repository.