ham-radio-software/D-Rats #218: Italian localization is not shown

maurizioandreotti, 2023-06-22:
as I always use software with english, i am sorry i hadn't noted this before.

In 0.4 version I setup language to be "italian" and i keep seeing the interface with english texts. (even closing/opening d-rats makes no difference)

Notes:
btw I did a small review of both the italian localization file and english (as this was empty) and regenerated the compilaed file and uploaded it into the github - I will do a PR for this revised versions

The regenerated .mo file works well if inserted into the 0.3.9 exe folder

maurizioandreotti, 2023-06-27:
In general all localized texts are not shown.

I guess it depends by the new UI which doesn't support these methods:

            #Gtk.glade.bindtextdomain("D-RATS", localedir)
            #Gtk.glade.textdomain("D-RATS")      

The only exception is the "info" box shown on Windows (on Mac it gets out in English):

[image]

maurizioandreotti, 2023-06-27:
@wb8tyw  I got to a stuck point with this... I was hoping to find some working example of a GTK3/python localized program, but so far nothing able to fix the labels on the GTK UI

wb8tyw, 2023-06-28:
It has been a source of frustration for me to find good python GTK-3 examples for every part of D-Rats.  Especially since there should be a lot of programs that are obviously working.

The only thing I can think of is that it should be using the current locale setting for the operating system as a default, so might not be able to be set from program control.

My assumption though is that you would have already set the locale setting, so I do not know what is wrong.

The only thing I can think of is try to write a "hello world" type simple programs for GTK-3 using both using only the API and with Glade XML template and see if they behave differently.

Perhaps it might be better to not use glade, and just use the GTK API.  That conversion would take a lot of work.

closed 2023-10-21 by wb8tyw