ham-radio-software/D-Rats #224: error adding a new email account

maurizioandreotti, 2023-07-09:
adding a new email account on mac (not yet tested on windows) I get:

`07/09/2023 09:33:04:INFO:D-Rats:---- GUI Exception ----
%s
---- End ----
Stack (most recent call last):
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-version/./d-rats.py", line 250, in <module>
    main()
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-version/./d-rats.py", line 244, in main
    app.main()
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-version/d_rats/mainapp.py", line 2136, in main
    self.run(None)
  File "/Users/maurizio/Virtualenvs/d-rats4/lib/python3.11/site-packages/gi/overrides/Gio.py", line 42, in run
    return Gio.Application.run(self, *args, **kwargs)
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-version/d_rats/mainwindow.py", line 291, in _activate_prefs
    saved = self._config.show(parent=self._window)
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-version/d_rats/config.py", line 2656, in show
    result = drats_ui.run()
  File "/Users/maurizio/Virtualenvs/d-rats4/lib/python3.11/site-packages/gi/overrides/Gtk.py", line 585, in run
    return Gtk.Dialog.run(self, *args, **kwargs)
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-version/./d-rats.py", line 94, in handle_exception
    MODULE_LOGGER.info("---- GUI Exception ----\n%s\n---- End ----\n",
07/09/2023 09:33:04:INFO:D-Rats:---- GUI Exception ----
%s
---- End ----
Stack (most recent call last):
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-version/./d-rats.py", line 250, in <module>
    main()
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-version/./d-rats.py", line 244, in main
    app.main()
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-version/d_rats/mainapp.py", line 2136, in main
    self.run(None)
  File "/Users/maurizio/Virtualenvs/d-rats4/lib/python3.11/site-packages/gi/overrides/Gio.py", line 42, in run
    return Gio.Application.run(self, *args, **kwargs)
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-version/d_rats/mainwindow.py", line 291, in _activate_prefs
    saved = self._config.show(parent=self._window)
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-version/d_rats/config.py", line 2656, in show
    result = drats_ui.run()
  File "/Users/maurizio/Virtualenvs/d-rats4/lib/python3.11/site-packages/gi/overrides/Gtk.py", line 585, in run
    return Gtk.Dialog.run(self, *args, **kwargs)
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-version/./d-rats.py", line 116, in handle_exception
    response = utils.make_error_dialog(msg, trace,
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-version/d_rats/utils.py", line 546, in make_error_dialog
    result = dialog.run()
  File "/Users/maurizio/Virtualenvs/d-rats4/lib/python3.11/site-packages/gi/overrides/Gtk.py", line 585, in run
    return Gtk.Dialog.run(self, *args, **kwargs)
  File "/Users/maurizio/Virtualenvs/d-rats4/D-Rats-version/./d-rats.py", line 94, in handle_exception
    MODULE_LOGGER.info("---- GUI Exception ----\n%s\n---- End ----\n",`

maurizioandreotti, 2023-07-09:
testing the same (adding new email account on mac) with PR 223 it prooduce a crash (seg-fault)

wb8tyw, 2023-07-16:
My test got:
`Traceback (most recent call last):
  File "/mnt/aviary/home/malmberg/work/d-rats/D-Rats/d_rats/keyedlistwidget.py", line 369, in render_password
    rend.set_property("text", "*" * len(val))
TypeError: object of type 'GdkFrameClockIdle' does not have property 'text'`

wb8tyw, 2023-07-16:
Related to https://github.com/ham-radio-software/D-Rats/issues/177

closed 2023-08-12 by wb8tyw

wb8tyw, 2023-08-12:
Some deprecated GTK code was found related to this and fixed.