ham-radio-software/D-Rats #228: Message with attachment does not validate on decompress

wb8tyw, 2023-07-16:
Message with an attachment does not validate when put into the outbox and attempted to be sent.

`msgfn <class 'str'> /Users/rww/.d-rats-ev/messages/Outbox/form_07162023_165529.xml
07/16/2023 17:09:10:INFO:MessageRouter:_run_one: broad-except
Traceback (most recent call last):
  File "/Users/rww/drats_07_2023/D-Rats/d_rats/msgrouting.py", line 750, in _run_one
    routed = self._route_message(msg, slist, routes)
  File "/Users/rww/drats_07_2023/D-Rats/d_rats/msgrouting.py", line 715, in _route_message
    routed = self._route_via_wl2k(src, route, msg)
  File "/Users/rww/drats_07_2023/D-Rats/d_rats/msgrouting.py", line 670, in _route_via_wl2k
    msg = self._form_to_wl2k_em(addr, msgfn)
  File "/Users/rww/drats_07_2023/D-Rats/d_rats/msgrouting.py", line 568, in _form_to_wl2k_em
    data = form.get_attachment(name)
  File "/Users/rww/drats_07_2023/D-Rats/d_rats/formgui.py", line 1378, in get_attachment
    return zlib.decompress(data)
zlib.error: Error -3 while decompressing data: incorrect header check
`

wb8tyw, 2023-07-16:
Not expecting it to be doing a decompress on a send operation, would expect it to be doing a send.

wb8tyw, 2023-07-17:
Draft was sent, nothing unusuall logged on sending side.
Receiving side did not get the attachment.
Got a traceback pm the receiving side about Session 4 ended but not registered.
line 123 sessions/control.py session= self._sm.session[ident]

Unfortunately this is one of the protocols that I have have not fully mapped out yet.

wb8tyw, 2023-07-17:
Looking back at the first traceback, it was to be sent over w2lk.

Since w2lk uses lzhuf conpression, it needs to gunzip the attachments before lzhuf compressing them, or the size of the attachment can grow.

I am not setup up to send/receive w2lk.  I used a dummy test server for unit tests, and will need to rig up something for testing this.

dranch, 2023-07-17:
Applications should not modify attachments.  If the file is a ZIP and the resulting transfer could be slower because the file is already compressed, either a) ideally D-Rats should be aware of specific file extensions (.zip, tgz, .jpg, gif, xz, etc) via some user configurable deny-list and NOT invoke lzhuf or just make lzhuf run every time and educate the user that attaching already compressed files will be slower

wb8tyw, 2023-07-17:
Well I am still trying to get a complete state diagram and or a flow chart on how the messaging works.

Some things are event driven, and some are timer driven, and in a few places if you try to send something to your self, it drops the message into a zombie in processing state forever until you do a manual cleanup of some of d-rats undocumented locking files.

D-Rats uses gzip compression on file transfers between clients. and ail messages are pretty much treated as file transfers.  Most of the code is common to both.

D-Rats pretty much is ignorant of mime-type and file extensions like Linux is, and code should never trust a file extension.

The D-Rats winlink message handling compresses all outgoing messages with lzhuf and expects to receive lzhuf compressed messages.

I am not aware of any options for controlling winlinks use of lzhuf.  lzhuf is pretty much unique to winlink, as until the patents expired a few years ago, it was not legal use use lzhuf compression with out purchasing a license in most of the world.   The Winlink support was added to D-Rats after the lzhuf compression patents expired in the U.S., Europe and Japan.

wb8tyw, 2023-07-20:
Have not found why attachments are not working in this case, but finding several issues in the form handling code addressing should help with figuring out what is going wrong.

As indicated in [Forms and messages-internal formats](https://github.com/wb8tyw/D-Rats/wiki/D-Rats-Forms-and-messages-internal-formats) the attachment file contents are stored as a gzipped string that is base64 encoded and inserted into the document.

closed 2023-08-12 by wb8tyw