ham-radio-software/D-Rats #230: Console gets Invalid DPRS symbol message every three seconds

dslotter, 2023-07-25:
Version 0.4.1dev Console log (NOT the event log) gets filled up with these repetitive messages and a google search doesn't reveal how to correct:

```
07/24/2023 21:03:05:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:08:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:11:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:14:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:17:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:20:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:23:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:26:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:29:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:32:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:35:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:38:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:41:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:44:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:47:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:50:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:53:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:56:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:03:59:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:04:02:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:04:05:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
07/24/2023 21:04:08:INFO:gps:dprs_to_aprs: Invalid DPRS symbol: `'
```

Running on Ubuntu Linux. Connected to gwinnettares.ratflector.net port 9000

wb8tyw, 2023-07-26:
The message is that somewhere you have a string that D-Rats thinks should contain a DPRS symbol code, but does not have a symbol code.

And as it does not give you a clue as to what the message is, that makes it a useless message and one that annoyingly floods the log.

As the message indicate, it is in the d_rats/gps.py module, and for a short term can be commented out.

The routine emitting the message is called in two places.  In one place it is when processing an NMEA GPS position and in the other in mainapp._refresh_location().  Since it is repeating, it looks like the mainapp call.

The mainapp call is trying to parse the default_gps_comment from the settings.

1. Config module allows the creation of an invalid default GPS comment or starts out with one.
Note that even if the config module is fixed, a manual edit to the file could create an invalid message, so probably the code needs to also deal with that.
2. The dprs_to_aprs() routine should not log this issue, it should throw an exception and the caller should catch that exception and write out a more useful log message if appropriate.

wb8tyw, 2023-08-13:
This is more of a mess that I realized.
The "default_gps_comment" is expected to by the code that is printing out the console log messages is expected to be just a two letter valid "DPRS" code.  I can not find a document for them anywhere.  D-Rats has internal methods to covert from APRS to DPRS.

It is obviously wrong for the the code to be interpreting the first two characters of the message to be this code.
There is also a DPRS button that can be used to select an APRS code and icon, with an optional message.  What it puts in the message is not the DPRS code based on the APRS code.

And then there is the box above it for putting in the two parts of the APRS code, instead of using the selection dialog that is used for the DPRS block to just let you select an based on an icon or the two letter code.

wb8tyw, 2023-08-19:
Starting to figure things out.
The last three characters, if they start with a asterisk '*', the next two are a checksum.
If that is found, the first two characters are a DPRS Symbol code, and the third character is an "overlay" code.
The overlay code appears to either a space or an underscore character.
The DPRS button will create or edit message that is compliant, but does not handle maintaining the overlay code assuming it must be a space in order to parse the existing text as a DPRS message.

closed 2023-09-18 by wb8tyw