ham-radio-software/D-Rats #232: Feature Request: Files transfers activity should be logged at repeater.log

KP4AJ, 2023-08-06:
Did some test, and when verifying repeater.log file transfer not recorded at repeater.log .  As repeater is the channel that facilitates the communication, IMO file activity should be recorded. It could be very informative and helpful in case of misuse. Also will help monitor bandwidth impact and more.  

Thanks for all the work with D-Rats! 

Edfel

KP4AJ, 2023-08-06:
File transfer at repeter.log could be similar to the one registered at the client side.  File 'X ' Sent.  With the identification of the sending and receiving stations.  Also, I observed the repeater.log record 'Delivering frame to station. '  IF record is made after a transfer is complete, repeater.log size get also reduced.   

Thanks.

Edfel

wb8tyw, 2023-08-07:
All transfers over a d-rats repeater are via a string of octets.
All the repeater does is take a packet in from one source, and then copies it to the outgoing port.

Files are broken up into multiple packets.  The repeater does not know how to decode those packets to determine what is being sent through it.

It would be a significant change to the repeater code to implement this.

KP4AJ, 2023-08-07:
Thanks for the explanation John!