ham-radio-software/D-Rats #233: remote file transfers enabled when not?

KP4AJ, 2023-08-07:
Hi:
If I understand functionality correctly leaving Preferences > Radio > Transfers > Remote file transfers _unmark_, should disable file transfers.  

Have made test with d-rats 3.10 beta 5 connect (Windows) and d-rats0.4.1.dev0 (Ubuntu) both versions accept file transfers with the Remote File Transfers setting Un-enabled.

To Reproduce.  Identify a station under Stations column, activate right click menu and select Send FIle.  Select file to be send.  File is sent.  

If I am mistaken the Remote files transfers setting, then there should be one to allow / disallow to receive files (client side). 

Edfel

KP4AJ, 2023-08-07:
Hi:
Ok noticed the setting seems to apply for downloads. I get the remote file transfer not enabled when trying a download. 

So to apply to Uploads, if was not the purpose of the setting for both (Downloads / Uploads), this could then be a feature request. 

Thanks.

KP4AJ, 2023-08-07:
Doing some more testing it seems something strange at File Upload logic. I mean A can send a file to B without being connected (I mean at the Files Menu). From my point of view that should be a bug, if others can reproduce. 

Also A sending a file without B allowing should not function that way. But I think this was not foresee at D-Rats original functionality with files.

wb8tyw, 2023-08-12:
Sending with out allowing receiving is a valid configuration.

Allowing receiving is giving permissions for others with out any secure validation to drop files on your device via D-Rats.
Marking this as an Enhancement for the configuration page.

Adding a setting to prevent by default to allow you to send may be a valid enhancement, but anyone using the D-Rats program could just go in and re-enable it, as we do not put a password on changing the configuration for D-Rats, and the D-Rats configuration file can be edited by any plain text editor program.

KP4AJ, 2023-08-13:
Well I have this idea. Station B connect to Station A.  Station A don't want to allow uploads. Lets create some kind of 'semaphore' file named notransfers to be placed at the files directory of Station A.   Operator of Station A manually creates the notransfers file with permission that do not allows it to be deleted. 

When Station B tries to upload and (D-Rats) read the directory a code (to be added to maybe main_files.py or session/file.py) that detect the presence of notransfers file automatically don't allow uploads and place a message at the files menu for example 'No Transfers Allowed'. Instead of FIle Sent, 

Again this is a concept, maybe other files have to be changed.  But I think could add a lot of security to D-Rats, forbidding files uploads when a station want to disable to receive them.  Hope the concept is stated clearly.

What I saw at some files, D-Rats client do read the remote station directory to create a list of files. That mechanism, IMO, could be enhanced to make room for this king of setup. A reading of a notransfers file within the files directory is a STOP.  

Looking forward, IF this could be implemented a second future improvement could be another 'semaphore'  file that states which file types are allowed for upload. But that is more long term. 

73'

Edfel
KP4AJ

KP4AJ, 2023-08-13:
In summary the task could be to create code with the functionality to detect a notransfers file at the remote station files directory. If notransfers is present  D-Rats will automatically treat the signal as Red Light or  'cancel request'.  File upload do' not proceed and message 'No Upload Transfers Allowed' is placed at the FIles Menu.  

Maybe this could be done with other mechanism, but I am looking into similar and existent mechanism that could be enhanced.  Making this, I hope,  feasible within a reasonable time frame and development effort. 

Edfel
KP4AJ

wb8tyw, 2023-08-13:
Any solution would be to be backwards compatible with older D-Rats clients.  Only the Python 3 version of D-Rats is planned for getting updates.