ham-radio-software/D-Rats #235: Callsign not following accepted formats

KP4AJ, 2023-08-12:
Hi:
Not functional related, but included here for a future.  Test set a callsign as ROBERT which not follows accepted format. Aware there are other key developments in process.  Also aware there are various callsign structure conventions across the world.  

[image]

73'

Edfel
KP4AJ

KP4AJ, 2023-08-12:
Well thinking of Tactical callsigns (Hospital, Distribution, etc) in EMCOMM situations then the callsign format flexibility could be justified.

wb8tyw, 2023-08-12:
There is an unused module in D-Rats that attempts to validate call signs for two countries.

There is also an unused structure in D-Rats that is being loaded with received call signs that I can not find anything ever reading.  So there is a lot of room for improvement

Related tickets: [Email gateway validation requires call to include dash suffix](https://github.com/ham-radio-software/D-Rats/issues/140)

I can not find the ticket, but it is possible that APRS and Winlink may have more restrictive requirements for the Station Session ID than what D-Rats does, However they all may share the same call sign and session id configuration.