ham-radio-software/D-Rats #239: Map navigation behavior issues?

flyingfrawg, 2023-08-28:
Updated my install this AM to test. File download works. No active users online to test msg and chat functions with, but did review map functions which are still a bit irregular. I appologize if these issues are described in previous reports.

Issue: when changing center positions of map, then changing zoom level, the map snaps back to the default position (my qth) rather than the new map center. I seem to be missing a step to force a new center; a single click on the desired new center does not work. Sometimes multiple (2 or more clicks) will force the new desired center position, but doing this seems inconsistent. 
When I can get a new desired center to "behave", it jumps around with increases and decreases in zoom level and requires repositioning with each zoom step. Multi-step zoom-in attempts often result in the desired location going off screen, sometimes not in the available map - so it can't be accessed by using the sliders.
I hope this report is helpful when the map functions are re-examined.
Is there a description of the desired behavior as now programmed?
It seems like this behavior should be made consistent.
-KD4IZ

wb8tyw, 2023-09-04:
Related ticket: https://github.com/ham-radio-software/D-Rats/issues/133

wb8tyw, 2023-09-14:
The mapping code is supposed to remember the latitude and longitude of the center of the map, which defaults to your station GPS coordinates if known.  For some currently unknown reason it takes two clicks on center here to actually change the center.

The actual calculation is to find the tile with the center of the map and then attempt to put that tile in the center of the map.  This is actually a complex calculation for what is called "slippy tiles".   Because of that at different zoom levels and the need place map tiles on the display window coordinates with floating to integer conversions it becomes hard to make sure that the scroll bars keep the current center with.  Keep in mind that the scroll bars do not change the center of the map, so scroll bar changes may be lost with the re-calculation for a zoom.

Rapid changes to zoom especially when tiles are not cached for that zoom level, or the cache is expired can result in overloading your computer with background loading of tiles, as it puts all requests for map tiles in their own thread with nothing limiting how many concurrent download threads, or canceling download requests for tiles that are no longer needed because the zoom level changes.  If you get too many tiles downloading, it can lockup D-Rats or crash it.