ham-radio-software/D-Rats #240: Unable to connect (Conversation error) Auth Enabled

KP4AJ, 2023-09-12:
Hi:
I am running a repeater only  setup (Ubuntu + PI 4).  I was able to modify the repeater python code to allow setup custom config directory.  Reference:
https://groups.io/g/d-rats/message/2672?p=%2C%2C%2C20%2C0%2C0%2C0%3A%3ACreated%2C%2Chack+repeater%2C20%2C2%2C0%2C100473319

D-Rats runs nicely BUT IF I enable Auth I get the error: Unable to connect (Conversation error) with Auth Enabled at config file.  A login is not completed.   Without Auth True connections ok and D-Rats functionality seems ok. 

Paths are as follow
/opt/d-rats/repeater.config

users.txt file gets read if I place at 
/opt/d-rats/

Otherwise I get a users.txt not found error at repeater log

Repeater log file gets placed at:
/home/drats/.d-rats.ev/repeater.log

I am interested in using D-Rats with Auth enabled.  Any idea what the error means and which files I should look into will be appreciated. 

Just placing here for feedback.

Thanks in advance.

73

Edfel
KP4AJ

KP4AJ, 2023-09-12:
systemctl status drats repeater shows:

 File "/home/dir/D-Rats/d_rats/transport.py", line 400, in auth_connect
Sep 12 02:48:02 host python3.11[31309]:     if authfn and not
authfn(self.pipe):
Sep 12 02:48:02 host python3.11[31309]: ^^^^^^^^^^^^^^^^^
Sep 12 02:48:02 host python3.11[31309]:   File
"/home/dir/D-Rats/d-rats_repeater.py", line 335, in auth_user
Sep 12 02:48:02 host python3.11[31309]:     username, password =
self.auth_exchange(pipe)
Sep 12 02:48:02 host python3.11[31309]: ^^^^^^^^^^^^^^^^^^^^^^^^
Sep 12 02:48:02 host python3.11[31309]:   File
"/home/dir/D-Rats/d-rats_repeater.py", line 299, in auth_exchange
Sep 12 02:48:02 host python3.11[31309]:     pipe.write(b"102 %s
okay\r\n" % cmd)
Sep 12 02:48:02 host python3.11[31309]: ~~~~~~~~~~~~~~~~~~~^~~~~
Sep 12 02:48:02 host python3.11[31309]: TypeError: %b requires a
bytes-like object, or an object that implements __bytes__, not 'str'

Using D-Rats 0.4.1 with the version.py fix.  Nit sure IF this error applies or affects standard D-Rats repeater use. 

Thanks again.

Edfel
KP4AJ

KP4AJ, 2023-09-12:
Hi Again:
Well seems some problem with binary and string data but that is beyond my capabilities...  > d-rats_repeater.py have not looked the other files. Or maybe something related with Python 3 migration.  

Well, just trying to figure out. 

Edfel

KP4AJ, 2023-09-13:
Hi John:
Thank you very much!!  Found the discussion about auth at: https://github.com/ham-radio-software/D-Rats/issues/201

Updated the D-Rats code with latest at git and applied my little hack to the d-rats_repeater.py and seems to be working!! with Auth = True!!  Errors gone.  Please mark this as SOLVED or whatever you consider appropiate!

Again, thank you very much John!!

73'

Edfel
KP4AJ

closed 2023-09-13 by KP4AJ

wb8tyw, 2023-09-14:
For your other issues, the d-rats repeater program already has command line options to tell it to run in a non-GUI environment, redirect the configuration file and log files with out changing the source code.

The current code in master will test to see if it has access to a display and if not takes the option for running with out a GUI.  I think that auto-detection requires the GTK python packages to be installed'.

References:
https://github.com/ham-radio-software/D-Rats/issues/241
https://github.com/ham-radio-software/D-Rats/issues/242

KP4AJ, 2023-09-14:
Excellent John!! Appreciated! Will test and report back. 

73'
Edfel
KP4AJ