ham-radio-software/D-Rats #243: D-Rats Repeater Errors

riggs498, 2023-09-18:
Here are my most recent d-rats repeater errors. I am not able to connect to anything remotely. I can connect to it. But it can not connect.

riggs498@virtualdesktop:~/D-Rats$ ./d-rats_repeater.py 
/home/riggs498/.local/lib/python3.10/site-packages/pydub/utils.py:170: RuntimeWarning: Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work
  warn("Couldn't find ffmpeg or avconv - defaulting to ffmpeg, but may not work", RuntimeWarning)
Traceback (most recent call last):
  File "/home/riggs498/D-Rats/./d-rats_repeater.py", line 1095, in button_on
    self.save_config(self.config)
  File "/home/riggs498/D-Rats/./d-rats_repeater.py", line 665, in save_config
    self.sync_config()
  File "/home/riggs498/D-Rats/./d-rats_repeater.py", line 1071, in sync_config
    self.config.set("settings", "devices", devices)
  File "/usr/lib/python3.10/configparser.py", line 1205, in set
    self._validate_value_types(option=option, value=value)
  File "/usr/lib/python3.10/configparser.py", line 1190, in _validate_value_types
    raise TypeError("option values must be strings")
TypeError: option values must be strings
Exception in thread Thread-2 (_repeat):
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/home/riggs498/D-Rats/./d-rats_repeater.py", line 419, in _repeat
    self.accept_new_gps()
  File "/home/riggs498/D-Rats/./d-rats_repeater.py", line 391, in accept_new_gps
    addr_str = "%s:$i" % addr
TypeError: not all arguments converted during string formatting

wb8tyw, 2023-09-18:
A temporary workaround is to use a text editor to edit the repeater configuration.

What is going on is that the python2 config library automatically converted data types to string.  The python3 config library does not.  And this was missed during testing.

riggs498, 2023-09-18:
Okay. I will do that.

Thanks,

Ryan (KB8PMY)
N8MFN Hamilton Wireless Radio Club

riggs498, 2023-09-18:
John,

No glory on trying that. I have attached my config file and the error. The
config file shows correctly in the program. But once you press on. It has a
bunch of nah I don't like you. LOL

[settings]
devices = [('net:10.0.2.7:9000',''), ('net:141.148.63.71:9000','')]
acceptnet = True
netport = 9000
id = N8MFN-A
idfreq = 30
require_auth = False
trust_local = True
gpsport = 9500
state = False

[tweaks]
allow_gps =

Traceback (most recent call last):
  File "/home/riggs498/.d-rats-ev/../D-Rats/d-rats_repeater.py", line 1095,
in button_on
    self.save_config(self.config)
  File "/home/riggs498/.d-rats-ev/../D-Rats/d-rats_repeater.py", line 665,
in save_config
    self.sync_config()
  File "/home/riggs498/.d-rats-ev/../D-Rats/d-rats_repeater.py", line 1071,
in sync_config
    self.config.set("settings", "devices", devices)
  File "/usr/lib/python3.10/configparser.py", line 1205, in set
    self._validate_value_types(option=option, value=value)
  File "/usr/lib/python3.10/configparser.py", line 1190, in
_validate_value_types
    raise TypeError("option values must be strings")
TypeError: option values must be strings

Thanks,

Ryan (KB8PMY)
N8MFN Hamilton Wireless Radio Club

wb8tyw, 2023-09-19:
Line 1071 in d-rats.repeater.py needs to be changed `to:
`        self.config.set("settings", "devices", str(devices))`

The opposite of most of the problems previously seen, all data must now be converted to strings before saving.

This will be fixed in a future patch / release of D-Rats.

closed 2023-09-22 by wb8tyw