ham-radio-software/D-Rats #248: pylint3 E Unable to locate package

KP4AJ, 2023-09-24:
Hi:
Doing a fresh Ubuntu 22.04 install I am facing trouble with pylint3:
E Unable to locate package

Found this explanation: 
https://community.nxp.com/t5/i-MX-Processors/pylint3/td-p/1513152

Will update this when been able to install D-Rats (Ubuntu 22.03) again.  BTW, was able to run D-Rats latest code from Ubuntu but wanted to do from fresh installation.  If pylint functionality for Python 3 is now supported by pylint, think documentation should remove pylint3 from packages required.  If there is other solution please share.   

I have latest code D-Rats code running on Windows 10. 

73'

Edfel
KP4AJ

wb8tyw, 2023-09-24:
Each distribution even Debian can name their packages differently.

My development of D-Rats is on msys2 and Anti-X Linux 22, and as such, only the development packages for Anti-X linux are currently documented.  The README.md file does state that, and that it states that the package list for other distributions may be different.

As distributions remove Python2 from their set of packages, some of them are also removing the "3" suffix from their packages instead of maintaining both the "3"  package and the unnumbered package.

The pylint package is not needed for running D-Rats, it is used only for doing some checks on the python files for developers.

Currently I do not have the time to setup the chroot / mocked builds for the D-Rats to have RPM / Debian as I have done for the LZHUF, and I do not have the time to setup chroot builds for LZHUF for Raspbian.

The WIKI for this project currently also needs some help as that is where this should be documented, again we do not have active volunteers in this project to help maintain the WIKI.

wb8tyw, 2023-09-24:
Until we get some more active volunteers on this project, this will not get worked on.