ham-radio-software/D-Rats #249: Error getting D-Rats to first install - GUI Exception

KP4AJ, 2023-09-24:
Hi:
Being trying since last night.  Get this:
09/24/2023 11:35:47:INFO:D-Rats:main: re-config option found -- Reconfigure D-rats
09/24/2023 11:35:47:INFO:D-Rats:main: args.config = /root/.d-rats-ev
09/24/2023 11:35:48:INFO:DratsConfig:File /root/.d-rats-ev/d-rats.config
09/24/2023 11:35:48:INFO:DratsConfig:/root/D-RATS Shared
09/24/2023 11:35:48:INFO:MainApp:_refresh_lang: Loading locale `en'
09/24/2023 11:35:48:INFO:MainApp:_refresh_lang: Locale dir is: /home/D-Rats/locale
09/24/2023 11:35:48:ERROR:MainApp:_refresh_lang: Messages catalog file missing  for en.  Need to use 'msgfmt tool to generate.
09/24/2023 11:35:48:INFO:MainApp:D-RATS v0.4.1.dev0 starting at Sun Sep 24 11:35:48 2023 on Platform UnixPlatform:
  configuration:       /root/.d-rats-ev
  system_data: /home/D-Rats
  OS version: Linux - Ubuntu 22.04.3 LTS
09/24/2023 11:35:50:INFO:D-Rats:---- GUI Exception ----
%s
---- End ----
Stack (most recent call last):
  File "/home/D-Rats/d-rats.py", line 94, in handle_exception
    MODULE_LOGGER.info("---- GUI Exception ----\n%s\n---- End ----\n",

Probably a missing library but no idea.  Will appreciate any pointer.  Calling D-Rats:
#sudo /usr/bin/python3 /home/D-Rats/d-rats.py   
.d-rats-ev shows only a maps directory but no config.

Thanks.

Edfel
KP4AJ

wb8tyw, 2023-09-24:
The Linux install is similar to what is done on Microsoft Windows after Msys2, a Linux emulation product is installed.

[Installation on Microsoft Windows](https://github.com/ham-radio-software/D-Rats/wiki/010.010-Installation-of-D-Rats-on-Microsoft-Windows) Which states if you install from the git repository that the message catalog needs to be manually built and shows a simple script to rebuild it.

The [README.md](https://github.com/ham-radio-software/D-Rats/blob/master/README.md) file in Master also specifies that the message catalog must be built, and references the WIKI [Translating-D-Rats-in-your-language](https://github.com/ham-radio-software/D-Rats/wiki/280---Translating-D-Rats-in-your-language) for a more complex way of creating the message catalog that the simple script that was developed since that Wiki page was written as part of creating python packaging.

The binary message catalog is intentionally not installed in the D-Rats repository.

wb8tyw, 2023-09-24:
Unfortunately it seems that the instructions to run build_pot.sh are not sufficient to generate needed mo files.
The only way to create the mo files currently is to run the python_prebuild.py, which requires having a virtualenv setup as per in the [README.md](https://github.com/ham-radio-software/D-Rats/blob/master/README.md)

wb8tyw, 2023-09-24:
Fixed build_pot.sh for future PR:

```bash
#!/bin/bash

set -uex

# First create the base.pot file
xgettext -d D-RATS -o locale/base.pot -D . *.py

# Next directory - Because of an odd string in ax25 need to
# specify a --from-code
xgettext -d D-RATS -o locale/base.pot -j --from-code ISO-8859-15 d_rats/*.py
# Repeat for each directory
# Any warnings probably need to be looked at.
xgettext -d D-RATS -o locale/base.pot -j d_rats/map/*.py
xgettext -d D-RATS -o locale/base.pot -j d_rats/sessions/*.py
xgettext -d D-RATS -o locale/base.pot -j d_rats/ui/*.py
xgettext -d D-RATS -o locale/base.pot -j -L Glade ui/*.glade

for file_name in $(find locale -name '*.po'); do
    locale_dir="$(dirname "$file_name")"
    msgfmt -o "$locale_dir/D-RATS.mo" "$locale_dir/D-RATS"
done
```

KP4AJ, 2023-09-24:
Hi John:

Thanks.  Here the result:
dminuser@emcommpr:/installs/GitHub/D-Rats$ sudo ./build_pot.sh
+ xgettext -d D-RATS -o locale/base.pot -D . d-rats.py d-rats_repeater.py internationalization_test.py plugin_test.py python_prebuild.py setup.py
+ xgettext -d D-RATS -o locale/base.pot -j --from-code ISO-8859-15 d_rats/agw.py d_rats/aprs_dprs.py d_rats/aprs_icons.py d_rats/ax25.py d_rats/callsigns.py d_rats/cap.py d_rats/comm.py d_rats/config.py d_rats/config_tips.py d_rats/crc_checksum.py d_rats/ddt2.py d_rats/debug.py d_rats/dplatform_generic.py d_rats/dplatform_macos.py d_rats/dplatform.py d_rats/dplatform_unix.py d_rats/dplatform_win32.py d_rats/dratsexception.py d_rats/emailgw.py d_rats/filenamebox.py d_rats/formbuilder.py d_rats/formgui.py d_rats/geocode_ui.py d_rats/gps.py d_rats/image.py d_rats/__init__.py d_rats/inputdialog.py d_rats/keyedlistwidget.py d_rats/latlonentry.py d_rats/listwidget.py d_rats/lzhuf.py d_rats/mailsrv.py d_rats/mainapp.py d_rats/mainwindow.py d_rats/map_source_editor.py d_rats/map_sources.py d_rats/menu_helpers.py d_rats/miscwidgets.py d_rats/msgrouting.py d_rats/pluginsrv.py d_rats/progressdialog.py d_rats/qst.py d_rats/reqobject.py d_rats/session_coordinator.py d_rats/sessionmgr.py d_rats/signals.py d_rats/spell.py d_rats/station_status.py d_rats/transport.py d_rats/treewidget.py d_rats/utils.py d_rats/version.py d_rats/wl2k.py d_rats/wu.py d_rats/yencode.py
+ xgettext -d D-RATS -o locale/base.pot -j d_rats/map/__init__.py d_rats/map/mapbottompanel.py d_rats/map/mapdisplay.py d_rats/map/mapdraw.py d_rats/map/mapexception.py d_rats/map/mapmarkerlist.py d_rats/map/mapmenumodel.py d_rats/map/mappopupmodel.py d_rats/map/mapposition.py d_rats/map/mapstatusbox.py d_rats/map/maptile.py d_rats/map/mapwidget.py d_rats/map/mapwindow.py d_rats/map/mapzoomcontrols.py d_rats/map/markereditdialog.py d_rats/map/markerpopupmodel.py
+ xgettext -d D-RATS -o locale/base.pot -j d_rats/sessions/base.py d_rats/sessions/chat.py d_rats/sessions/control.py d_rats/sessions/file.py d_rats/sessions/form.py d_rats/sessions/__init__.py d_rats/sessions/rpc.py d_rats/sessions/sniff.py d_rats/sessions/sock.py d_rats/sessions/stateful.py d_rats/sessions/stateless.py
+ xgettext -d D-RATS -o locale/base.pot -j d_rats/ui/account_dialog.py d_rats/ui/conntest.py d_rats/ui/event_popup_model.py d_rats/ui/event_tab.py d_rats/ui/__init__.py d_rats/ui/main_chat.py d_rats/ui/main_common.py d_rats/ui/main_events.py d_rats/ui/main_files.py d_rats/ui/main_messages.py d_rats/ui/main_stations.py d_rats/ui/message_folder_info.py d_rats/ui/message_folders.py d_rats/ui/message_list.py d_rats/ui/message_popup_model.py d_rats/ui/station_popup_model.py d_rats/ui/test_message_popup_model.py
+ xgettext -d D-RATS -o locale/base.pot -j -L Glade ui/addport.glade ui/mainwindow.glade
++ find locale -name '*.po'
+ for file_name in $(find locale -name '*.po')
++ dirname locale/de/LC_MESSAGES/D-RATS.po
+ locale_dir=locale/de/LC_MESSAGES
+ msgfmt -o locale/de/LC_MESSAGES/D-RATS.mo locale/de/LC_MESSAGES/D-RATS
+ for file_name in $(find locale -name '*.po')
++ dirname locale/nl/LC_MESSAGES/D-RATS.po
+ locale_dir=locale/nl/LC_MESSAGES
+ msgfmt -o locale/nl/LC_MESSAGES/D-RATS.mo locale/nl/LC_MESSAGES/D-RATS
+ for file_name in $(find locale -name '*.po')
++ dirname locale/es/LC_MESSAGES/D-RATS.po
+ locale_dir=locale/es/LC_MESSAGES
+ msgfmt -o locale/es/LC_MESSAGES/D-RATS.mo locale/es/LC_MESSAGES/D-RATS
+ for file_name in $(find locale -name '*.po')
++ dirname locale/it/LC_MESSAGES/D-RATS.po
+ locale_dir=locale/it/LC_MESSAGES
+ msgfmt -o locale/it/LC_MESSAGES/D-RATS.mo locale/it/LC_MESSAGES/D-RATS
+ for file_name in $(find locale -name '*.po')
++ dirname locale/en/LC_MESSAGES/D-RATS.po
+ locale_dir=locale/en/LC_MESSAGES
+ msgfmt -o locale/en/LC_MESSAGES/D-RATS.mo locale/en/LC_MESSAGES/D-RATS
adminuser@emcommpr:/installs/GitHub/D-Rats$ cd locale
adminuser@emcommpr:/installs/GitHub/D-Rats/locale$ ls
base.pot  de  en  es  it  nl
adminuser@emcommpr:/installs/GitHub/D-Rats/locale$ cd en
adminuser@emcommpr:/installs/GitHub/D-Rats/locale/en$ ls
LC_MESSAGES
adminuser@emcommpr:/installs/GitHub/D-Rats/locale/en$ cd LC_MESSAGES
adminuser@emcommpr:/installs/GitHub/D-Rats/locale/en/LC_MESSAGES$ ls
D-RATS.mo  D-RATS.po
adminuser@emcommpr:/installs/GitHub/D-Rats/locale/en/LC_MESSAGES$ ls -
ls: cannot access '-': No such file or directory
adminuser@emcommpr:/installs/GitHub/D-Rats/locale/en/LC_MESSAGES$ ls -o
total 88
-rw-r--r-- 1 root 29470 sep 24 18:10 D-RATS.mo
-rw-r--r-- 1 root 56906 sep 24 18:06 D-RATS.po
adminuser@emcommpr:/installs/GitHub/D-Rats/locale/en/LC_MESSAGES$ 

The .mo file got created.  However D-Rats still refuses to start.

dminuser@emcommpr:/installs/GitHub/D-Rats$ sudo ./d-rats.py
09/24/2023 18:20:48:INFO:D-Rats:main: re-config option found -- Reconfigure D-rats
09/24/2023 18:20:48:INFO:D-Rats:main: args.config = /root/.d-rats-ev
09/24/2023 18:20:48:INFO:DratsConfig:File /root/.d-rats-ev/d-rats.config
09/24/2023 18:20:48:INFO:DratsConfig:/root/D-RATS Shared
09/24/2023 18:20:48:INFO:MainApp:_refresh_lang: Loading locale `en'
09/24/2023 18:20:48:INFO:MainApp:_refresh_lang: Locale dir is: /installs/GitHub/D-Rats/locale
09/24/2023 18:20:48:INFO:MainApp:D-RATS v0.4.1.dev17-ga0b0c90-dirty starting at Sun Sep 24 18:20:48 2023 on Platform UnixPlatform:
  configuration:       /root/.d-rats-ev
  system_data: /installs/GitHub/D-Rats
  OS version: Linux - Ubuntu 22.04.3 LTS
09/24/2023 18:20:54:INFO:D-Rats:---- GUI Exception ----
%s
[... cut ...]
09/24/2023 18:22:22:INFO:MainApp:D-RATS v0.4.1.dev17-ga0b0c90-dirty starting at Sun Sep 24 18:22:22 2023 on Platform UnixPlatform:
  configuration:       /root/.d-rats-ev
  system_data: /installs/GitHub/D-Rats
  OS version: Linux - Ubuntu 22.04.3 LTS
09/24/2023 18:22:26:INFO:D-Rats:---- GUI Exception ----
%s
---- End ----
Stack (most recent call last):
  File "/installs/GitHub/D-Rats/d-rats.py", line 94, in handle_exception
    MODULE_LOGGER.info("---- GUI Exception ----\n%s\n---- End ----\n",
adminuser@emcommpr:/installs/GitHub/D-Rats$ 

Thanks for your help!

Edfel

wb8tyw, 2023-09-25:
```text
09/24/2023 18:20:48:INFO:MainApp:_refresh_lang: Locale dir is: /installs/GitHub/D-Rats/locale
09/24/2023 18:20:48:INFO:MainApp:D-RATS v0.4.1.dev17-ga0b0c90-dirty starting at Sun Sep 24 18:20:48 2023 on Platform UnixPlatform:
configuration: /root/.d-rats-ev
system_data: /installs/GitHub/D-Rats
```

Unfortunately the feature of D-Rats of putting up the GUI exception means that the information needed to diaglose the problem is only in the GUI popup in a section that must b expanded.   None of that critical information makes it into the message log.

Nothing in d-rats or the d-rats repeater requires root privileges.
It is an extremely unsafe practice to use the root account for anything other than system maintenance.
Running d-rats or d-rats_repeater from root will not be allowed in a future update to d-rats.

KP4AJ, 2023-09-25:
Roger that John!  Thanks for looking at this.  Will re-check my setup.  Appreciated.

73'

Edfel
KP4AJ

KP4AJ, 2023-09-25:
John:
Not sure if this could help:
Traceback (most recent call last):

  File "/installs/GitHub/D-Rats/./d-rats.py", line 250, in <module>
    main()

  File "/installs/GitHub/D-Rats/./d-rats.py", line 228, in main
    app = mainapp.MainApp(safe=args.safe)

  File "/installs/GitHub/D-Rats/d_rats/mainapp.py", line 380, in __init__
    if not self.config.show():

  File "/installs/GitHub/D-Rats/d_rats/config.py", line 2720, in show
    drats_ui = DratsConfigUI(self, parent)

  File "/installs/GitHub/D-Rats/d_rats/config.py", line 2530, in __init__
    self.build_ui()

  File "/installs/GitHub/D-Rats/d_rats/config.py", line 2616, in build_ui
    add_panel(DratsGPSPanel, "gps", _("GPS config"), prefs, self)

  File "/installs/GitHub/D-Rats/d_rats/config.py", line 2591, in add_panel
    panel = c_arg(self.config, *args)

  File "/installs/GitHub/D-Rats/d_rats/config.py", line 1403, in __init__
    dprs_code = dprs_info['code']

KeyError: 'code'

73'

Edfel

KP4AJ, 2023-09-25:
John:
Changed folder permissions and executed from user 'adminuser' (no sudo).  Per the build_pot.sh fix  .mo file was created and at locale/en/LC_MESSAGES dir:
adminuser@emcommpr:/home/D-Rats$ ./d-rats.py
09/25/2023 12:44:32:INFO:D-Rats:main: re-config option found -- Reconfigure D-rats
09/25/2023 12:44:32:INFO:D-Rats:main: args.config = /home/adminuser/.d-rats-ev
09/25/2023 12:44:32:INFO:DratsConfig:File /home/adminuser/.d-rats-ev/d-rats.config
09/25/2023 12:44:32:INFO:DratsConfig:/home/adminuser/D-RATS Shared
09/25/2023 12:44:32:INFO:MainApp:_refresh_lang: Loading locale `en'
09/25/2023 12:44:32:INFO:MainApp:_refresh_lang: Locale dir is: /home/D-Rats/locale
09/25/2023 12:44:32:INFO:MainApp:D-RATS v0.4.1.dev17 starting at Mon Sep 25 12:44:32 2023 on Platform UnixPlatform:
  configuration:       /home/adminuser/.d-rats-ev
  system_data: /home/D-Rats
  OS version: Linux - Ubuntu 22.04.3 LTS
09/25/2023 12:44:36:INFO:D-Rats:---- GUI Exception ----
%s
---- End ----
Stack (most recent call last):
  File "/home/D-Rats/./d-rats.py", line 94, in handle_exception
    MODULE_LOGGER.info("---- GUI Exception ----\n%s\n---- End ----\n",

This is a Ubuntu Laptop with GtK3 installed.   

Edfel

wb8tyw, 2023-09-26:
Click ignore the ignore or similar button on the error popup when it comes up, it is not fatal, and is only present at the current tip of the master branch.

Then go into the Preferences menu, GPS Config, and Edit a new Default GPS Comment.  The edit button forces the Default GPS comment to be in a valid format now.   In the past the default value of the Default GPS comment was not a valid GPS comment.

This will be fixed in a future PR.

KP4AJ, 2023-09-26:
John, it causes a crash, program closes after pressing ignore or ignore all.   No access to go into Preferences.  Maybe a manual cp of a config file into the directory could work. But my interest is help make it work without manual work around.

Very appreciated your work with D-Rats.

Edfel
KP4AJ

wb8tyw, 2023-09-27:
This is from a different branch than master, so the line numbers might not be right.  I is unlikely that I will get the pull request that will include this fix ready before at least next week.
```text
diff --git a/d_rats/config.py b/d_rats/config.py
index 9fd5b73..ee287f1 100755
--- a/d_rats/config.py
+++ b/d_rats/config.py
@@ -1386,23 +1386,27 @@ class DratsGPSPanel(DratsPanel):
                 config.set("settings", "default_gps_comment", dprs)
                 val.child_widget.set_text(dprs)
                 dprs_info = APRSicons.parse_dprs_message(text=dprs)
-                dprs_code = dprs_info['code']
-                if 'overlay' in dprs_info:
-                    dprs_code = dprs_info['code'] + dprs_info['overlay']
-                aprs_code = AprsDprsCodes.dprs_to_aprs(
-                    code=dprs_code,
-                    default=AprsDprsCodes.APRS_FALLBACK_CODE)
-                pixbuf = APRSicons.get_icon(code=aprs_code)
-                dprs_icon.set_from_pixbuf(pixbuf)
+                if 'code' in dprs_info:
+                    dprs_code = dprs_info['code']
+                    if 'overlay' in dprs_info:
+                        dprs_code = dprs_info['code'] + dprs_info['overlay']
+                    aprs_code = AprsDprsCodes.dprs_to_aprs(
+                        code=dprs_code,
+                        default=AprsDprsCodes.APRS_FALLBACK_CODE)
+                    pixbuf = APRSicons.get_icon(code=aprs_code)
+                    dprs_icon.set_from_pixbuf(pixbuf)
 
         val = DratsConfigWidget(config, "settings", "default_gps_comment")
         val.add_text(20)
         val.set_sensitive(False)
         dprs_comment = config.get("settings", "default_gps_comment")
         dprs_info = APRSicons.parse_dprs_message(text=dprs_comment)
-        dprs_code = dprs_info['code']
-        if 'overlay' in dprs_info:
-            dprs_code = dprs_info['code'] + dprs_info['overlay']
+        if 'code' in dprs_info:
+            dprs_code = dprs_info['code']
+            if 'overlay' in dprs_info:
diff --git a/d_rats/config.py b/d_rats/config.py
index 9fd5b73..ee287f1 100755
--- a/d_rats/config.py
+++ b/d_rats/config.py
@@ -1386,23 +1386,27 @@ class DratsGPSPanel(DratsPanel):
                 config.set("settings", "default_gps_comment", dprs)
                 val.child_widget.set_text(dprs)
                 dprs_info = APRSicons.parse_dprs_message(text=dprs)
-                dprs_code = dprs_info['code']
-                if 'overlay' in dprs_info:
-                    dprs_code = dprs_info['code'] + dprs_info['overlay']
-                aprs_code = AprsDprsCodes.dprs_to_aprs(
-                    code=dprs_code,
-                    default=AprsDprsCodes.APRS_FALLBACK_CODE)
-                pixbuf = APRSicons.get_icon(code=aprs_code)
-                dprs_icon.set_from_pixbuf(pixbuf)
+                if 'code' in dprs_info:
+                    dprs_code = dprs_info['code']
+                    if 'overlay' in dprs_info:
+                        dprs_code = dprs_info['code'] + dprs_info['overlay']
+                    aprs_code = AprsDprsCodes.dprs_to_aprs(
+                        code=dprs_code,
+                        default=AprsDprsCodes.APRS_FALLBACK_CODE)
+                    pixbuf = APRSicons.get_icon(code=aprs_code)
+                    dprs_icon.set_from_pixbuf(pixbuf)
 
         val = DratsConfigWidget(config, "settings", "default_gps_comment")
         val.add_text(20)
         val.set_sensitive(False)
         dprs_comment = config.get("settings", "default_gps_comment")
         dprs_info = APRSicons.parse_dprs_message(text=dprs_comment)
-        dprs_code = dprs_info['code']
-        if 'overlay' in dprs_info:
-            dprs_code = dprs_info['code'] + dprs_info['overlay']
+        if 'code' in dprs_info:
+            dprs_code = dprs_info['code']
+            if 'overlay' in dprs_info:
+            if 'overlay' in dprs_info:
+                dprs_code = dprs_info['code'] + dprs_info['overlay']
+        else:
+            dprs_code = '  '
         aprs_code = AprsDprsCodes.dprs_to_aprs(
             code=dprs_code,
             default=AprsDprsCodes.APRS_FALLBACK_CODE)
```

```text
$ git diff d_rats/mainapp.py
diff --git a/d_rats/mainapp.py b/d_rats/mainapp.py
index bb8de57..7a80cf2 100755
--- a/d_rats/mainapp.py
+++ b/d_rats/mainapp.py
@@ -1186,7 +1186,7 @@ class MainApp(Gtk.Application):
                 fix.aprs_code = AprsDprsCodes.dprs_to_aprs(
                     code=dprs_code)
                 self.default_comment = comment
-        except (NoOptionError, DPRSInvalidCode):
+        except (KeyError, NoOptionError, DPRSInvalidCode):
             # silently fix this up.  Notifications here at info level will
             # flood the console log.
             self.logger.debug("_refresh_location comment='%s'",
```

There will also be a patch to change the default string to one that will pass the checksum but this code just falls back to making up its own string if the value in the configuration file is bad quietly, which is pretty close the what the old behavior was.

KP4AJ, 2023-09-27:
Thanks John will test tomorrow after returning home from work.

Edfel
KP4AJ

closed 2023-10-21 by wb8tyw