ham-radio-software/D-Rats #252: Download issue (on radio )

maurizioandreotti, 2023-10-07:
Using file download over the radio, pushing "download" button I got this error:
______________
Traceback (most recent call last):

  File "C:/msys64\home\mauri\D-Rats-version/d_rats/mainapp.py", line 1471, in __submit_rpc_job
    self.rpc_session(port).submit(job)

  File "C:/msys64\home\mauri\D-Rats-version/d_rats/sessions/rpc.py", line 726, in submit
    self.__send_job(job, ident)

  File "C:/msys64\home\mauri\D-Rats-version/d_rats/sessions/rpc.py", line 700, in __send_job
    frame = self.__job_to_frame(job, ident)

  File "C:/msys64\home\mauri\D-Rats-version/d_rats/sessions/rpc.py", line 639, in __job_to_frame
    frame.data = self.__encode_rpccall(job)

  File "C:/msys64\home\mauri\D-Rats-version/d_rats/sessions/rpc.py", line 629, in __encode_rpccall
    return "%s%s%s" % (job.__class__.__name__, ASCII_GS, job.pack())

  File "C:/msys64\home\mauri\D-Rats-version/d_rats/sessions/rpc.py", line 237, in pack
    return encode_dict(self._args)

  File "C:/msys64\home\mauri\D-Rats-version/d_rats/sessions/rpc.py", line 112, in encode_dict
    value = value.decode('utf-8', 'replace')

AttributeError: 'NoneType' object has no attribute 'decode'
________________

Traceback (most recent call last):
  File "C:/msys64\home\mauri\D-Rats-version/d_rats/ddt2.py", line 338, in unpack
    h_index = val.index(ENCODED_HEADER) + len(ENCODED_TRAILER)
ValueError: subsection not found
10/07/2023 14:37:15:INFO:SessionManager:incoming: Received block 0:0 for session `chat'
10/07/2023 14:37:15:INFO:SessionManager:incoming: Received block 5:1 for session `rpc'
10/07/2023 14:37:17:INFO:RPCSession:Sending job `Request file None' to IZ2FOW
10/07/2023 14:37:17:INFO:D-Rats:---- GUI Exception ----
%s
---- End ----
Stack (most recent call last):
  File "C:\msys64\home\mauri\D-Rats-version\d-rats.py", line 250, in <module>
    main()
  File "C:\msys64\home\mauri\D-Rats-version\d-rats.py", line 244, in main
    app.main()
  File "C:/msys64\home\mauri\D-Rats-version/d_rats/mainapp.py", line 2144, in main
    self.run(None)
  File "C:/msys64/mingw64/lib/python3.10/site-packages/gi/overrides/Gio.py", line 42, in run
    return Gio.Application.run(self, *args, **kwargs)
  File "C:/msys64\home\mauri\D-Rats-version/d_rats/ui/main_files.py", line 437, in _download
    self.emit("submit-rpc-job", job, port)
  File "C:\msys64\home\mauri\D-Rats-version\d-rats.py", line 94, in handle_exception
    MODULE_LOGGER.info("---- GUI Exception ----\n%s\n---- End ----\n",
10/07/2023 14:37:19:INFO:SessionManager:incoming:Received frame for station `!'
10/07/2023 14:37:19:INFO:SessionManager:incoming: Received block 7:1 for session `rpc'
10/07/2023 14:37:40:INFO:RPCSession:worker: Cancelling job 6 due to timeout
10/07/2023 14:37:40:INFO:RemoteFileV:_file_list_cb : Incomplete job
10/07/2023 14:37:54:INFO:DDT2EncodedFrame:unpack: Block has no header/trailer
Traceback (most recent call last):
  File "C:/msys64\home\mauri\D-Rats-version/d_rats/ddt2.py", line 338, in unpack
    h_index = val.index(ENCODED_HEADER) + len(ENCODED_TRAILER)
ValueError: subsection not found
10/07/2023 14:37:54:INFO:SessionManager:incoming: Received block 0:0 for session `chat'
10/07/2023 14:37:54:INFO:DDT2EncodedFrame:unpack: Checksum failed: 1170 != 36670
10/07/2023 14:37:54:INFO:SessionManager:incoming: Received block 0:0 for session `chat'
10/07/2023 14:38:00:INFO:SessionManager:incoming: Received block 0:0 for session `chat'

wb8tyw, 2023-10-07:
Can you duplicate this issue with a ratflector connection?

wb8tyw, 2023-10-07:
Download failed over the serial line for me with a number of errors:
```text
  File "/usr/lib/python3.9/logging/__init__.py", line 363, in getMessage
    msg = msg % self.args
TypeError: %x format: an integer is required, not bytes

  File "/mnt/aviary/home/malmberg/work/d-rats/D-Rats/d_rats/comm.py", line 374, in is_xon
    self.logger.info("is_xon: Aiee! Read a non-XOFF char: 0x%02x `%s`",
Message: 'is_xon: Aiee! Read a non-XOFF char: 0x%02x `%s`'
Arguments: (b'[', b'[')
10/07/2023 18:47:35:INFO:SWFSerial:is_xon: Assuming IXANY behavior

  File "/mnt/aviary/home/malmberg/work/d-rats/D-Rats/d_rats/comm.py", line 374, in is_xon
    self.logger.info("is_xon: Aiee! Read a non-XOFF char: 0x%02x `%s`",
Message: 'is_xon: Aiee! Read a non-XOFF char: 0x%02x `%s`'
Arguments: (b'[', b'[')
10/07/2023 18:47:35:INFO:SWFSerial:is_xon: Assuming IXANY behavior

```

wb8tyw, 2023-10-08:
Probably related to https://github.com/ham-radio-software/D-Rats/issues/253

D-rats serial code for 0.3 and 0.4 may randomly discard a byte received from the serial port.